OpenC3/cosmos #1139: Third party CI/CD pipeline interface

drthomso, 2024-03-11:
Hi there,

Please excuse me if a solution already exists for the question I'm about to ask. I tried in vain to find one.

Is there a way to have a CI/CD pipeline access COSMOS and run the tests scripts that are written there and then consume the results as text or some other way?

For example, I have a Gitlab runner that I use to test software on a device ignoring the COSMOS plugin and any COSMOS software, but if I want to run COSMOS tests written in Ruby etc. on that device I have to open COSMOS and manually run each in the script runner.

I investigated the JSON API and it didn't seem to suggest that it supports running existing scripts or sending an executable file and then consuming their output. 

Any help or suggestions would be greatly appreciated.

jmthomas, 2024-03-15:
You can use Playwright like we do to do full UI testing or run things headless and simply run scripts against the server. We don't have instructions for that yet but it will eventually show up in our Guides or Developer docs.

drthomso, 2024-03-18:
Thanks Jason. I appreciate you getting back to me so quickly.

tsjordan-eng, 2024-04-25:
We were able to get this working with `curl`ing the REST API in a bash script. We even convert the test output to a junit.xml to parse it later. Where do you want this documented?

ryanmelt, 2024-04-25:
If you are running it with Curl, then the curl page might make sense: https://github.com/OpenC3/cosmos/blob/main/docs.openc3.com/docs/development/curl.md

closed 2024-04-28 by ryanmelt