OpenC3/cosmos #1142: Python formatted does not work for get_target_file() read

timothyscherer, 2024-03-13:
I have a Python script that reads from a target's binary file. I tried to follow the instructions [here](https://docs.openc3.com/docs/guides/scripting-api#get_target_file) for Python, using something very similar:
```py
file = get_target_file(f"RCE/returns/capture/{name.lower()}_{channel}.bin")
print(formatted(file.read()))
```

This gets the following error:
```
2024-03-13T17:16:47.495877Z (generate_return_plots.py:12): {"time": 1.710350207495877e+18, "@timestamp": "2024-03-13T17:16:47.495877Z", "level": "ERROR", "microservice_name": "Script Runner", "detail": "generate_return_plots.py:12", "container_name": "14bd866567c3", "message": "Traceback (most recent call last):\n File \"RCE/lib/generate_return_plots.py\", line 12, in plot\n print(formatted(file.read()))\n ^^^^^^^^^^^^^^^^^^^^^^\n File \"/usr/lib/python3.11/site-packages/openc3/utilities/string.py\", line 66, in formatted\n string += \"%02X\" % byte\n ~~~~~~~^~~~~~\nTypeError: %X format: an integer is required, not str\n", "type": "log"}
```

I believe this has something to do with how the file is read out using `get_target_file()`, maybe [this](https://github.com/OpenC3/cosmos/blob/e555b5ba4127590bcedf79edf486babb8829869e/openc3/python/openc3/script/storage.py#L108)? Opening the file in text mode might be causing unexpected behavior.

OpenC3 COSMOS Version 5.15.0

closed 2024-03-15 by jmthomas

timothyscherer, 2024-03-18:
@jmthomas it seems like this may not be fixed still. Now, my script fails on `get_target_file()`.

Here is function call:
```py
def plot(name, channel):
  file = get_target_file(f"RCE/public/capture/{name.lower()}_{channel}.bin")
```

Here is the result:
[image]

Here is the output:
```log
2024-03-18T20:23:51.827819Z (generate_return_plots.py:9): {"time": 1.710793431827819e+18, "@timestamp": "2024-03-18T20:23:51.827819Z", "level": "ERROR", "microservice_name": "Script Runner", "detail": "generate_return_plots.py:9", "container_name": "78cdef9dbe1c", "message": "Traceback (most recent call last):
File \"RCE/lib/generate_return_plots.py\", line 9, in plot
file = get_target_file(f\"RCE/public/capture/{name.lower()}_{channel}.bin\")
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File \"/usr/lib/python3.11/site-packages/openc3/script/storage.py\", line 118, in get_target_file
raise error
File \"/usr/lib/python3.11/site-packages/openc3/script/storage.py\", line 112, in get_target_file
return _get_storage_file(f\"{part}/{path}\", scope=scope)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File \"/usr/lib/python3.11/site-packages/openc3/script/storage.py\", line 162, in _get_storage_file
file.write(response.text)
File \"/usr/lib/python3.11/tempfile.py\", line 500, in func_wrapper
return func(*args, **kwargs)
       ^^^^^^^^^^^^^^^^^^^^^
TypeError: a bytes-like object is required, not 'str'
", "type": "log"}
```

ryanmelt, 2024-03-19:
It wasn't fully fixed yet.  I finished it in #1148.
Will be in the next release.

PR #1143 (2024-03-15): Fix download_file and get_target_file
closes #1142