OpenC3/cosmos #1144: Missing Tlm packets in CmdTlmServer and Packet Viewer after upgrading from 5.14.2 to 5.15.0

jwalters2, 2024-03-14:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
After upgrading my Cosmos installation from 5.14.2 to 5.15.0, only 5 out of my 9 defined telemetry packets are showing in CmdTlmServer under the Tlm packets tab and also on Packet Viewer.

The plugin output in the Admin Console shows all of the telemetry packet definitions being processed and all of the telemetry is viewable in Telemetry Viewer.  Telemetry Viewer is unable to create a new screen from the packets that are "missing".

Verified that the same plugin GEM file installed properly on Cosmos 5.14.2.  All defined telemetry packets are seen.

**To Reproduce**
Install GEM plugin on Cosmos 5.15.0

**Expected behavior**
All defined packets are visible in all locations.

**Environment (please complete the following information):**

- OS: Mac OS 13.6.5
- Docker Desktop: 4.28.0
- OpenC3 COSMOS Version 5.15.0
- Browser: Chrome

Also verified same behavior on Ubuntu running Docker using a Chrome browswer.

ryanmelt, 2024-03-14:
Are the missing packets flagged as HIDDEN?  That was fixed in COSMOS 5.15.0

jwalters2, 2024-03-14:
The packets had telemetry items that were to be hidden.  Didn't realize that the keyword was applied to the whole packet.  The HIDDEN keyword was removed and the packets showed up.  Thanks.

closed 2024-03-14 by jwalters2

ryanmelt, 2024-03-14:
See IGNORE_ITEM in target.txt for hiding individual items.   We have an open ticket to be able to do that at the individual packet item level like you were trying to do with the HIDDEN keyword, but yes, HIDDEN is a packet level setting.