OpenC3/cosmos #1147: build tools failed

sensationHJ, 2024-03-15:
I followed this Tool Generator [page](https://docs.openc3.com/docs/getting-started/generators#tool-generator) 

after execute `openc3.sh cli generate tool DataVis` ,

I checked 
```
TOOL datavis "DataVis"
  INLINE_URL js/app.js
  ICON mdi-file-cad-box
```
is in plugin.txt.

and I didn't edit any files that generated automatically.

but I faced this error

```
cd <plugin folder>
../openc3.sh cli rake build VERSION=1.0.2 --trace
** Invoke build (first_time)
** Invoke require_version (first_time)
** Execute require_version
** Execute build
yarn run build
rake aborted!
yarn run build failed
/openc3/local/Rakefile:21:in `block (2 levels) in <top (required)>'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/file_utils.rb:57:in `sh'
/openc3/local/Rakefile:20:in `block in <top (required)>'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:281:in `block in execute'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:281:in `each'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:281:in `execute'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:199:in `synchronize'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:199:in `invoke_with_call_chain'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/task.rb:188:in `invoke'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:182:in `invoke_task'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:138:in `block (2 levels) in top_level'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:138:in `each'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:138:in `block in top_level'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:147:in `run_with_threads'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:132:in `top_level'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:83:in `block in run'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:208:in `standard_exception_handling'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/lib/rake/application.rb:80:in `run'
/usr/lib/ruby/gems/3.2.0/gems/rake-13.1.0/exe/rake:27:in `<top (required)>'
/usr/bin/rake:25:in `load'
/usr/bin/rake:25:in `<main>'
Tasks: TOP => build
```

how can i fix this error?

thanks

jmthomas, 2024-03-15:
I can reproduce but I'm not sure the root cause. Open the `Rakefile` and delete the `sh('yarn', 'run', 'build')` line. Then manually type `yarn`, then `yarn run build`, then `../openc3.sh cli rake build VERSION=1.0.2`

sensationHJ, 2024-03-19:
thanks to reply! 

I figured out that and edit Rakefile you mentioned and edit openc3.sh
```bash
cliroot_cmd )

    . "$(dirname -- "$0")/.env"
    args=`echo $@ | { read _ args; echo $args; }`
    (docker network create openc3-cosmos-network || true) &> /dev/null
    # docker run -it --rm --env-file "$(dirname -- "$0")/.env" --user=root --network openc3-cosmos-network -v `pwd`:/openc3/local:z -w /openc3/local $OPENC3_REGISTRY/$OPENC3_NAMESPACE/openc3-node$OPENC3_IMAGE_SUFFIX:$OPENC3_TAG yarn&&yarn run build
    docker run -it --rm --env-file "$(dirname -- "$0")/.env" --user=root --network openc3-cosmos-network -v `pwd`:/openc3/local:z -w /openc3/local $OPENC3_REGISTRY/$OPENC3_NAMESPACE/openc3-node$OPENC3_IMAGE_SUFFIX:$OPENC3_TAG $args
    set +a
    ;;
```
and run `../openc3.sh yarn` and `../openc3.sh yarn run build`

so I success to build !

[image]

thanks a lot!!!!!

closed 2024-03-22 by jmthomas