OpenC3/cosmos #1158: Suddenly no response from server

MightyArty, 2024-03-25:
Hey, I'm Tom from AST Space Mobile, and we have been using your library a lot lately.
I'm sending some commands using your library and all works fine, but sometimes without any known issue from my side, I get an exception raised on your side that is coming from this path: **openc3\script\server_proxy.py.**
The exception is the following:
`openc3.io.json_drb_object.JsonDRbError: No response from server: Request: {'url': 'http://127.0.0.1:2900/openc3-api/api', 'data': '{"jsonrpc": "2.0", "method": "tlm", "params": ["BLOCKONE", "MIC_LSL-MIC_FILE_INFO_RES", "RECEIVED_COUNT"], "keyword_params": {"scope": "DEFAULT"}, "id": 0}', 'headers': {'User-Agent': 'OpenC3 / v5 (ruby/openc3/lib/io/json_drb_object)', 'Content-Type': 'application/json-rpc', 'Authorization': 'admin'}} ::: None ::: Exception: ConnectionError(MaxRetryError("HTTPConnectionPool(host='127.0.0.1', port=2900): Max retries exceeded with url: /openc3-api/api (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x00000182CE9FB670>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it'))"))`.

Then in some cases, after waiting a couple of seconds (for example between 10-30 seconds) and sending again the same command packet without changing anything it suddenly works.

Could you please assist with this issue? I would be glad if I could catch this exception somehow from my side and manage it.

ryanmelt, 2024-03-25:
Hi Tom -
Just FYI: This is the kind of issue we would normally jump onto a support call with our customers.

The error you are getting "No connection could be made because the target machine actively refused it", is the typical error you see when a TCP server isn't listening.

If it repairs itself after a few seconds, that might indicate the openc3-cosmos-cmd-tlm-api container is restarting.

Can you run "docker ps" and see if any of your containers have restarted recently when this happens?
If so please do "docker logs <container hash>" and get some more information on what is going.

General problems that can cause containers to restart would be running out of RAM or disk space.

jmthomas, 2024-04-07:
Closing due to lack of new information

closed 2024-04-07 by jmthomas

MightyArty, 2024-04-07:
> Hi Tom - Just FYI: This is the kind of issue we would normally jump onto a support call with our customers.
> 
> The error you are getting "No connection could be made because the target machine actively refused it", is the typical error you see when a TCP server isn't listening.
> 
> If it repairs itself after a few seconds, that might indicate the openc3-cosmos-cmd-tlm-api container is restarting.
> 
> Can you run "docker ps" and see if any of your containers have restarted recently when this happens? If so please do "docker logs " and get some more information on what is going.
> 
> General problems that can cause containers to restart would be running out of RAM or disk space.

After an investigation of the Docker containers, it seems that the issue was with the openc3-cosnos-cmd-tlm-api container as you've mentioned.
I've added retry on this exception with some delay and it worked for me.