OpenC3/cosmos #1170: Bridge support installation fails over missing openssl; sys/socket.h

jurgensymynck, 2024-04-04:
Hi,

After quite some time I picked up working om my OpenC3 COSMOS project again...

I have a clean installed pc, Windows 11, 
I followed the correct steps involving cloning openc3-project repo, running the demo and can compile my plugins for OpenC3
I have Ruby 3.2 including full dev kit. **I installed this as admin, not as user**...
When running  `gem install openc3` as [instructed here](https://docs.openc3.com/docs/guides/bridges), the installation failed over missing `openssl.h` file.
An intermediate fix was `gem install openssl`, but ultimately the installer stops again over missing ´sys/socket.h´... no idea what gem I should install here to make further progress.

Any ideas? Thanks in advance!
Below is my (abriged) command line session in Windows Powershell (running as Admin), including a printout of  `gem environment`

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

PS C:\Windows\system32> gem install openc3
Fetching yard-0.9.36.gem
Fetching websocket-native-1.0.0.gem
Fetching websocket-1.2.10.gem
Fetching webrick-1.8.1.gem
Fetching uuidtools-2.2.0.gem
Fetching concurrent-ruby-1.2.3.gem
Fetching tzinfo-2.0.6.gem
Fetching tzinfo-data-1.2024.1.gem
Fetching raabro-1.4.0.gem
Fetching et-orbi-1.2.11.gem
Fetching fugit-1.10.1.gem
Fetching rufus-scheduler-3.9.1.gem
Fetching rubyzip-2.3.2.gem
Fetching connection_pool-2.4.1.gem
Fetching redis-client-0.21.1.gem
Fetching redis-5.1.0.gem
Fetching rack-3.0.10.gem
Fetching rackup-2.1.0.gem
Fetching nio4r-2.7.1.gem
Fetching puma-6.4.2.gem
Fetching opentelemetry-api-1.2.5.gem
Fetching opentelemetry-semantic_conventions-1.10.0.gem
Fetching opentelemetry-registry-0.3.1.gem
Fetching opentelemetry-common-0.20.1.gem
Fetching opentelemetry-sdk-1.4.1.gem
Fetching opentelemetry-instrumentation-base-0.22.3.gem
Fetching opentelemetry-instrumentation-redis-0.25.3.gem
Fetching opentelemetry-instrumentation-rack-0.24.0.gem
Fetching opentelemetry-instrumentation-faraday-0.24.1.gem
Fetching opentelemetry-instrumentation-aws_sdk-0.5.1.gem
Fetching opentelemetry-instrumentation-action_pack-0.9.0.gem
Fetching google-protobuf-3.25.3-x64-mingw-ucrt.gem
Fetching googleapis-common-protos-types-1.14.0.gem
Fetching opentelemetry-exporter-otlp-0.26.3.gem
Fetching nokogiri-1.16.3-x64-mingw-ucrt.gem
Fetching mqtt-0.6.0.gem
Fetching multi_json-1.15.0.gem
Fetching jsonpath-1.1.5.gem
Fetching httpclient-2.8.3.gem
Fetching hiredis-client-0.21.1.gem
Fetching ffi-1.16.3-x64-mingw-ucrt.gem
Fetching multipart-post-2.4.0.gem
Fetching faraday-multipart-1.0.4.gem
Fetching faraday-net_http-3.1.0.gem
Fetching faraday-2.9.0.gem
Fetching faraday-follow_redirects-0.3.0.gem
Fetching childprocess-4.1.0.gem
Fetching cbor-0.5.9.8.gem
Fetching openc3-5.15.2.gem
Fetching aws-eventstream-1.3.0.gem
Fetching aws-sigv4-1.8.0.gem
Fetching jmespath-1.6.2.gem
[... 144 lines cut ...]
     - C:\Ruby32\bin
     - C:\Python3\Scripts\
     - C:\Python3\
     - C:\Strawberry\c\bin
     - C:\Strawberry\perl\site\bin
     - C:\Strawberry\perl\bin
     - C:\gs10\bin
     - C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common
     - C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\
     - C:\Program Files\dotnet\
     - C:\Program Files\Microsoft VS Code\bin
     - C:\Git\cmd
     - C:\WinSCP\
     - C:\ffmpeg\bin
     - C:\putty\
     - C:\Program Files (x86)\GitExtensions\
     - C:\cygwin64\bin
     - C:\Program Files\F3D\bin
     - C:\Program Files\Docker\Docker\resources\bin
     - C:\openc3-project
     - C:\Users\XXXXXXXXXXXXXXXXXXX\AppData\Local\Microsoft\WindowsApps
     - C:\Users\XXXXXXXXXXXXXXXXXXX\AppData\Local\Programs\MiKTeX\miktex\bin\x64\
PS C:\Windows\system32>

```

ryanmelt, 2024-04-04:
I haven't tried to install the gem directly on Windows since we added hiredis in 5.15.0.  
You might need to go back to using version 5.14.2 of the gem on the host until then.
```gem install openc3 -v 5.14.2```

jmthomas, 2024-04-04:
You're supposed to be able to install hiredis in Windows and [this line](https://github.com/redis-rb/redis-client/blob/3c669f44c7478ffb222e483c8acbdb2a00bbfd65/hiredis-client/ext/redis_client/hiredis/extconf.rb#L11) won't compile on Windows.

Can you open `irb` and type: `RUBY_PLATFORM` and see what it says

jurgensymynck, 2024-04-05:
Thanks fior the quick response!
Here is the output:
```
irb(main):001:0> RUBY_PLATFORM
=> "x64-mingw-ucrt"
irb(main):002:0>
```

jurgensymynck, 2024-04-05:
> I haven't tried to install the gem directly on Windows since we added hiredis in 5.15.0. You might need to go back to using version 5.14.2 of the gem on the host until then. `gem install openc3 -v 5.14.2`

thanks for the workaround, I'm up and running!

greetings, Jurgen

jmthomas, 2024-04-05:
@jurgensymynck I just submitted a patch to the redis-client so when that gets fixed upstream this should resolve. Thanks for raising this!

PR #1199 (2024-04-12): Bump redis-client version
closes #1170
closed 2024-04-12 by jmthomas