OpenC3/cosmos #1193: Query on wait_check() api.

Lakshmi5785-dev, 2024-04-11:
Hi tream,
     I am trying to fetch a telemetry point value of a telemetry packet. Here I am using wait_check as below
wait_check(target, packet_name, telem_point, \"!= 0\", 5)

When I run the test for the first time this fails with value of telem point received as None.
When I just re-run the test it passes.
I increased the wait time but still it fails for the first time.This happens always.
Can you please help me to solve this.

jmthomas, 2024-04-12:
I can't reproduce:

`wait_check('INST', 'HEALTH_STATUS', 'TEMP1', '!= 0', 5)`

Works in our demo in both Ruby and Python.

Note that we recommend the alternative syntax of:

`wait_check("INST HEALTH_STATUS TEMP1 != 0", 5)`

ryanmelt, 2024-04-12:
Note you will receive None if that packet has never been received.

jmthomas, 2024-04-16:
Assuming the initial check was None due to packet not received. This is the expected behavior.

closed 2024-04-16 by jmthomas