OpenC3/cosmos #1195: Can you use Python interfaces with bridges/routers?

mkalin2, 2024-04-11:
The router documentation states that only Ruby interfaces can be used to instantiate a router interface, could the equivalent Python interface files be used instead?

- [image]

If not, can you at least specify a custom Python interface as your bridge interface when creating a bridge? The bridge documentation seems to imply that you can only run bridges with Ruby...

- [image]

jmthomas, 2024-04-12:
The Python implementation of bridges is not yet complete. I'll use this as a ticket for that implementation.

jmthomas, 2025-07-09:
Related #1079

PR #2244 (2025-09-09): [Feature] Python Serial Bridge
closes #1195 
closes #1079