OpenC3/cosmos #1243: Python Custom Interface Module Not Found Error

mkalin2, 2024-05-07:
I have a custom Python interface that I have tried putting in my plugin /lib directory as well as the top level openc3/python/interfaces directory. I can successfully build my plugin, but when I upload it to COSMOS the interface microservice crashes because it cannot find my custom interface.

```
['Traceback (most recent call last):\
', '  File \"/usr/lib/python3.11/site-packages/openc3/microservices/microservice.py\", line 48, in class_run\
    microservice = cls(name)\
                   ^^^^^^^^^\
', '  File \"/openc3/python/openc3/microservices/interface_microservice.py\", line 435, in __init__\
    ).build()\
      ^^^^^^^\
', '  File \"/usr/lib/python3.11/site-packages/openc3/models/interface_model.py\", line 147, in build\
    klass = get_class_from_module(\
            ^^^^^^^^^^^^^^^^^^^^^^\
', '  File \"/usr/lib/python3.11/site-packages/openc3/top_level.py\", line 155, in get_class_from_module\
    importlib.import_module(f\".{parts[-1]}\", \".\".join(parts[0:-1]))\
', '  File \"/usr/lib/python3.11/importlib/__init__.py\", line 126, in import_module\
    return _bootstrap._gcd_import(name[level:], package, level)\
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\
', '  File \"<frozen importlib._bootstrap>\", line 1204, in _gcd_import\
', '  File \"<frozen importlib._bootstrap>\", line 1176, in _find_and_load\
', '  File \"<frozen importlib._bootstrap>\", line 1140, in _find_and_load_unlocked\
', \"ModuleNotFoundError: No module named 'openc3.interfaces.custom_interface'\
\"]", "type": "log"}
```

Am I doing something wrong?

jmthomas, 2024-05-08:
What is the name of the file: `custom_interface.py`? How are you trying to use it? What is the INTERFACE line in your plugin.txt?

mkalin2, 2024-05-08:
The name of the file is custom_interface.py 

The plugin.txt file reads:
 `INTERFACE <%= scope_target_name %>_INT openc3/interfaces/custom_interface.py 128.244.216.61 1.0 None TERMINATED 0x0A 0X0A`

I was able to fix this issue, however, by running ./openc3.sh build and restarting COSMOS. Does that sound right?

jmthomas, 2024-05-08:
No the `build` command rebuilds all the containers. Did you clone the source (this repo) instead of the [project](https://github.com/OpenC3/cosmos-project/)? You should be able to build the plugin with the `custom_interface.py` file in the lib directory and then install with a plugin.txt like this:

```
INTERFACE <%= scope_target_name %>_INT custom_interface.py 128.244.216.61 1.0 None TERMINATED 0x0A 0X0A
```

mkalin2, 2024-05-09:
I have been working off the the cosmos repo rather than cosmos-project. I tried running the cosmos-project repo and building my plugin there, but I still get the same error when I have my interface in the lib folder and my plugin.txt as you describe.

jmthomas, 2024-07-13:
I believe this is working as intended in the cosmos-project. Please re-open if you can reproduce with a minimal example in the DEMO.

closed 2024-07-13 by jmthomas