OpenC3/cosmos #1265: Notification Limit Link Broken

tsjordan-eng, 2024-05-13:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
The link to the exceeded limit in the notification is broken

**To Reproduce**
Steps to reproduce the behavior:

1. Get an out of limits notification
2. Click the red bell in the upper right
3. Click your limit notification
4. Click Open button

**Expected behavior**
It should open either the limits tools or the packet viewer, but right now it just opens a new `about:blank` tab.

- OS: Windows 11
- OpenC3 COSMOS Version 5.16.1 (also in 5.15.2)
- Browser: Edge

tsjordan-eng, 2024-05-14:
Can I also just add here that the "clear notifications" icon is not very intuitive.  I always have to hover over the tool tip

jmthomas, 2024-05-21:
> Can I also just add here that the "clear notifications" icon is not very intuitive. I always have to hover over the tool tip

@tsjordan-eng we're using the Google material design icon which is literally called "notification-clear-all": https://pictogrammers.com/library/mdi/icon/notification-clear-all/

ryanmelt, 2024-05-21:
That icon is horrible though

tsjordan-eng, 2024-05-22:
Haha, yeah I noticed that's the icon in that space themed style guide y'all use. I really can't think of anything better to use. Android just has a button called "clear". IPhone seems to have an x button that conforms with a "clear" button. Windows has a "clear all" button. I can't think of a good icon and I guess MS, Apple, and Google can't either.

PR #1291 (2024-05-22): Notifications: Only open if url, change clear icon
closes #1265
closed 2024-05-22 by jmthomas