OpenC3/cosmos #1269: Update streaming api to support string values

sunws4me, 2024-05-15:
Streaming api only handles numeric values today.  would like support for strings as well.  This would allow EU strings (enums) and tlm points that are native strings to be displayed in grafana.

ryanmelt, 2024-05-15:
The streaming api does support non-numeric values (It is the API used by Data Extractor).   
Can you provide an example that isn't working?   (Is it the Ruby or Python version that isn't working and maybe just Javascript is working?)

jmthomas, 2024-05-15:
It's specifically the Grafana plugin that isn't interpreting the string data correctly. In the plugin we're doing this:
```
                  frame.addField({ name: item, type: FieldType.number });
```
I tried doing changing to `FieldType.string` but then that changes all the values to a string and nothing graphs. I'm not sure yet how to support both.