OpenC3/cosmos #1283: Elegant way of sharing constants definitions like ID_PARAMETER (packet ID's) in COSMOS and C++ header files?

jurgensymynck, 2024-05-20:
Hello,

I wonder if there is an elegant way to prevent defining constants like ID_PARAMETER (packet ID's) in multiple places, like in lower example?

The goal of course is to prevent 'magic numbers' in the targets configuration files, and specify constants in one single place; a file.

This file (what type and format?) should be accessed while building the COSMOS plugin, and also be used somehow with the C++ preprocessor to define constants.

I realize the best solution is to dynamically allocate all these values after the device has booted, connected to COSMOS, and have the variables set by a COSMOS command, but I want to keep the code complexity to a minimum at this time.

Any pointers in the right direction are welcome!
greetings, Jurgen

### in cmd.txt:
```
APPEND_ID_PARAMETER PACKET_ID 8 UINT 0x61 0x61 0x61
```

#### in a C++ header file (for AVR 2560):
```
const uint8_t CMD_SET_RATE = 0x61;
```

Same principal could be used to have shared safely limits for a device:

#### in tlm.txt
```
APPEND_ITEM SERVO_ANGLE_PULSE_VALUE 16 UINT "Joint/Servo [0]"
    LIMITS DEFAULT 1 ENABLED 110 111 524 525
```

#### in a C++ header file:
```
const servo_safety_limits_t servo_safety_limits[3] = {
    {110, 111, 524, 525},   // #  servo [0]
    {112, 113, 497, 498},   // #  servo [1]
    {108, 109, 502, 503}   // #  servo [2]
};
```

ryanmelt, 2024-05-20:
Hi @jurgensymynck,

Customers typically handle this with a 3rd "master database" that outputs both the COSMOS configuration and the C++ configuration.   (Or just accept having the same numbers in two places)

Another option is to have the COSMOS files as your master database, and then write a small ruby script that reads the COSMOS config files and then generates the C++ headers.

jurgensymynck, 2024-06-03:
thanks!

I will experiment with ruby erb templates ans see how far i get.

Jurgen

closed 2024-10-12 by jmthomas