OpenC3/cosmos #1297: Cosmos crushes after the modification of SAMPLE_TLM.txt file

lucapizzuto99, 2024-05-23:
Hi everyone,

I wanted to modify the telemetry file of the sample subsystem, in order to add new parameters.
At first, this seemed to work, since I was able to see the new parameter with its values in the packet viewer.
However, when I tried to modify the file to add other parameters, COsmos crushed and I was not able to display the interfaces. 

I then decided to report the file to its original state, without any modification, but that didn't work.

[image]

The error I get is:

**Environment (please complete the following information):**

- OS: [e.g. Windows 10, Mac OS 10.13]
- OpenC3 COSMOS Version [e.g. 5.0.6]
- Browser: [e.g. Chrome, Firefox, Safari, Edge]

jmthomas, 2024-05-23:
Try to load again. Sometimes you get an "already exists at create" message per #1271

lucapizzuto99, 2024-05-23:
Hi Thomas, thanks for the quick reply.
What I did multiple times is to reverse the changes done and go back to the functioning version of the repo of one week ago, then stop cosmos (and the entire NOS3) and then re-make.
However, the issue persists.
Is this what you meant by loading again?

jmthomas, 2024-05-23:
You might be using the old version of the Demo plugin if you edited it directly. You shouldn't be rebuilding the entire repo. Just rebuild the plugin `openc3.sh cli rake build VERSION=1.0.0` and then upload it (or update it) on the Admin screen.

lucapizzuto99, 2024-05-24:
Hi Thomas.
I actually have installed cosmos as part of the entire NOS3 simulator, so I am not sure if this applies to my situation, like I do not have the openc3.sh file in my project.
Also, the error message is this one:

`plugin.txt20240524-7-svw1xb:92: MAP_TARGET CFDP_TEST_RADIO
Error: Unknown keyword and parameters for Target: MAP_TARGET CFDP_TEST_RADIO
Backtrace:
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.12.0/lib/openc3/models/target_model.rb:558:in `handle_config'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.12.0/lib/openc3/models/plugin_model.rb:240:in `block in install_phase2'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.12.0/lib/openc3/config/config_parser.rb:217:in `parse_loop'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.12.0/lib/openc3/config/config_parser.rb:217:in `parse_file'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.12.0/lib/openc3/models/plugin_model.rb:226:in `install_phase2'
/openc3/bin/openc3cli:467:in `load_plugin'
/openc3/bin/openc3cli:649:in `<main>'
`
The runtime error of before is not the reason why cosmos crushes, I think it has to do with this error. 

My NOS3/gsw/cosmos folder is structured in this way:
[image]

I am sorry but I am still new to NOS3 and Openc3.

closed 2026-11-05 by EmilyRagan