OpenC3/cosmos #1304: ability to execute scripts uploaded by "put_target_file" outside script runner

321github123, 2024-05-29:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We would like to execute scripts created by "put_target_file" using the python api.

**Describe the solution you'd like**
We would like a python api call that achieves similar functionality as run_script in the timeline microservice i.e. execute a script that has been uploaded as if the script had been run by script runner

[image]

**Additional context**
Add any other context or screenshots about the feature request here.

ryanmelt, 2024-05-29:
We added the python api to spawn scripts in ScriptRunner in 5.16.2.   Jason should have a pypi package for that pushed today with the correct version number.  
https://github.com/OpenC3/cosmos/blob/main/openc3/python/openc3/script/script_runner.py#L70

jmthomas, 2024-05-29:
openc3 5.16.2 pushed to pypi: https://pypi.org/project/openc3/

However, I'm having issues running scripts due to an authentication error. Still trying to track it down.

321github123, 2024-05-30:
Is there anyway to programmatically login by chance?  We want to run in automated manner  "put_target_file" --> execute target file

ryanmelt, 2024-06-01:
There is a set_offline_access() api available in Ruby that we will need to make available in Python.
Both languages could use an easier way to generate one.  Right now would require low level HTTP requests to keycloak without any assistance.

PR #1851 (2025-04-03): Implement offline access in python
closes #1304 

I don't think this is complete to fully close 1304 but I did the initial port of the offline access stuff.
closed 2025-04-03 by ryanmelt