OpenC3/cosmos #1309: plugin.txt VARIABLE array options as combobox

jmthomas, 2024-05-31:
Currently the VARIABLE keyword takes two parameters, the name and the default value. I'd like it to take additional values which result in the VARIABLE being rendered as a combobox with only those options available.

`VARIABLE target_name INST INST2 INST3` would render a combobox with three choices: INST, INST2, INST3

If the user wanted to add an option that currently doesn't exist they can directly edit the plugin.txt so I don't think we need a custom input. Plus the plugin author is making an explicit choice that these are the only available options.

Note we also want to support a future VARIABLE description option as specified in #764. This would look something like:

`VARIABLE target_name INST INST2 INST3 "Select the target name"`

So we'll have to check for quotes on the last parameter when that is implemented.

ryanmelt, 2024-05-31:
Most things like this I would like to handle with additional keywords.   
There is a lot of detail we can add to the plugin install process.   For example the combobox options could be VARIABLE_STATE for example.   Trying to lean away from the really long unnamed parameter lists that we have used in the past.

jmthomas, 2024-05-31:
So then if additional OPTIONS are given we render combobox:
```
VARIABLE target_name INST
  OPTION INST2
  OPTION INST3
```

That would still allow for the description to be given on the same line ... that makes parsing easier and easier to edit.

ryanmelt, 2024-05-31:
I think everything associated with VARIABLE should start with VARIABLE_
Also the first given can be the assumed default.