OpenC3/cosmos #1314: Python Block Depth Limit of only 5

ikarlinsey, 2024-06-03:
**Description**
The typical block depth limit of python is 20. When executing scripts in script runner, the block depth limit is 5. 

**Steps to reproduce the behavior**
1. Go to Script Runner
2. Write the following script:

```python
a = True
b = False
c = False
d = False
e = False
f = False

if a:
  print ("a")
elif b:
  print ("b")
elif c:
  print ("c")
elif d:
  print ("d")
elif e:
  print ("e")
``` 

3. Execute the script, notice that it works successfully.
4. Add an additional elif to the script:
```python
a = True
b = False
c = False
d = False
e = False
f = False

if a:
  print ("a")
elif b:
  print ("b")
elif c:
  print ("c")
elif d:
  print ("d")
elif e:
  print ("e")
elif f:
  print ("f")
``` 
5. Execute the script, notice that it fails to execute with the following error:
SyntaxError: too many statically nested blocks

**Expected behavior**
Python standard block depth is 20 

**Environment (please complete the following information):**

- OS: Windows 11 Enterprise
- OpenC3 COSMOS Version 5.16.3-beta0 & OpenC3 Cosmos Version 5.14.2
- Browser: Chrome 125.0.6422.113

ryanmelt, 2024-06-03:
This is a side effect of us instrumenting the code.  (We wrap every line so we can show the file executing line by line and catch exceptions).

If you have large blocks like this you'll need to put them in uninstrumented code. (Other files that you import)

jmthomas, 2024-10-12:
Not sure what changed but this now works as expected:
```
a = False
b = False
c = False
d = False
e = False
f = False
g = False
h = False
i = False
j = False
k = False
l = False
m = False
n = False
o = False
p = False
q = False
r = False
s = False
t = False
u = False
v = False
w = False
x = False
y = False
z = True

if a:
  print ("a")
elif b:
  print ("b")
elif c:
  print ("c")
elif d:
  print ("d")
elif e:
  print ("e")
elif f:
  print ("f")
elif g:
  print ("g")
elif h:
  print ("h")
elif i:
  print ("i")
elif k:
  print ("k")
elif l:
  print ("l")
elif m:
  print ("m")
elif n:
  print ("n")
elif o:
  print ("o")
elif p:
  print ("p")
elif q:
  print ("q")
elif r:
  print ("r")
elif s:
  print ("s")
elif t:
  print ("t")
elif u:
  print ("u")
elif v:
  print ("v")
elif w:
  print ("w")
elif x:
  print ("x")
elif y:
  print ("y")
elif z:
  print ("z")
```

closed 2024-10-12 by jmthomas