OpenC3/cosmos #1316: Make build_cmd timeout a variable that can be passed into call

321github123, 2024-06-04:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
The build_cmd call occasionally times out. When we rerun once or twice, then it succeeds.

**Describe the solution you'd like**
In python/ruby build_cmd api call, expose a time out parameter with a default value.

**Additional context**
[image]

JoshuaNeely, 2024-06-04:
We manually increased timeout to 60s (building src code) and see the same intermittent error that made us think it was a timeout thing.

We might instead need to troubleshoot our redis connection... retry logic, etc. There's something else afoot

ryanmelt, 2024-06-04:
Check the docker logs on the operator to see if there are any errors there.

JoshuaNeely, 2024-06-04:
Nothing jumped out at me in the logs.

Current theory is our redis pod is maxing out it's memory allocation

jmthomas, 2024-07-13:
closed by #1381

closed 2024-07-13 by jmthomas