OpenC3/cosmos #1317: Discard leading bytes not working properly?

mkalin2, 2024-06-04:
I have an interface that is using the length protocol and the command/response protocol (however this also seems to happen even when I just use the length protocol by itself). **Everything works as expected when the "Discard Leading Bytes" field is set to 0.** 

Here is the log output when I send a command when discard leading bytes is nonzero: 
```
2024-06-04 12:32:21.257 | INFO | DEFAULT__INTERFACE__<my_device>_INT | <my_device>_INT: Connection Success
2024-06-04 12:32:21.253 | INFO | DEFAULT__INTERFACE__<my_device>_INT | <my_device>_INT: Connect 128.244.216.61:1861 (R/W)
2024-06-04 12:32:16.248 | INFO | DEFAULT__INTERFACE__<my_device>_INT | <my_device>_INT: Connection Lost
2024-06-04 12:32:16.247 | INFO | DEFAULT__INTERFACE__<my_device>_INT | <my_device>_INT: Internal disconnect requested (returned nil)
2024-06-04 12:32:16.247 | INFO | DEFAULT__INTERFACE__<my_device>_INT | <my_device>_INT: read_interface requested disconnect
2024-06-04 12:32:16.247 | INFO | DEFAULT__INTERFACE__<my_device>_INT | <my_device>_INT: OpenC3::TcpipClientStream read returned 0 bytes (stream closed)
``` 
For some reason it thinks there is an empty response.

Here is what my plugin.txt looks like:
```
TARGET <my_device> 
INTERFACE <my_device>_INT tcpip_client_interface.rb 128.244.216.61 1861 1861 10.0 None
  MAP_TARGET <my_device>
  PROTOCOL READ_WRITE LengthProtocol 32 32 8 1 BIG_ENDIAN 8 nil nil TRUE
  PROTOCOL READ_WRITE CmdResponseProtocol
```
Here are what cmd.txt and tlm.txt look like:
```
COMMAND <my_device> GET_ID BIG_ENDIAN
  APPEND_PARAMETER OPERATION 8 BLOCK 0x81
  APPEND_PARAMETER HEADER_VERSION 8 UINT 1 1 1
  APPEND_PARAMETER SEQUENCE_NUMBER 8 UINT 1 1 1
  APPEND_PARAMETER SPARE 8 UINT 0 0 0
  APPEND_PARAMETER CMD_LENGTH 32 UINT 5 5 5
  APPEND_ID_PARAMETER CMD 40 STRING "*IDN?"
  RESPONSE <my_device> ID_RESP

TELEMETRY <my_device> ID_RESP BIG_ENDIAN "Response to *IDN?"
  ACCESSOR TemplateAccessor
  TEMPLATE "<MANUFACTURER>,<PART_NUM>,<SERIAL_NUM>,<SOFTWARE_VERSION>"
  APPEND_ITEM MANUFACTURER 168 STRING
  APPEND_ITEM PART_NUM 168 STRING
  APPEND_ITEM SERIAL_NUM 168 STRING
  APPEND_ITEM SOFTWARE_VERSION 168 STRING
```

As you can see I am just sending simple SCPI commands, but there is an 8 byte overhead that I have to send and receive with each SCPI command that is specific to this device. There is a length field in that overhead which is why I am using the length protocol. Although I don't really need to interpret the length field in the response since I am using CmdResponse, but I have tried setting the read protocol to burst, and I still have the same issue when trying to discard leading bytes.

My goal with the discard leading bytes is to eliminate that overhead from the packet once it is displayed in COSMOS. It doesn't really matter to me if this elimination happens before or after the data gets filled in the template (if it gets eliminated after the templating happens, I will just need to account for those extra 8 bytes when setting the size of that field). 

Am I doing things correctly? Is this a bug with discard leading bytes? It seems to happen regardless of using whether or not I am using CommandResponse but maybe the template accessor is causing problems as well?

ryanmelt, 2024-06-04:
OpenC3::TcpipClientStream read returned 0 bytes (stream closed)

This usually means the other side closed the socket.   Is the SCPI device closing the socket because it is receiving badly formatted data?

ryanmelt, 2024-06-04:
It sounds like you don't want to discard leading bytes on the command correct?   Right now you are.

mkalin2, 2024-06-04:
I do not want to discard those bytes in the command, if they were being discarded that would definitely cause problems with the SCPI device...

I didn't realize the discard applies to commands as well. I can't imagine a scenario where you would want to delete any of the bytes _before_ they get sent out the interface, but thanks for clearing that up.

I updated my plugin.txt to only discard on read:

```
TARGET <my_device> 
INTERFACE <my_device>_INT tcpip_client_interface.rb 128.244.216.61 1861 1861 10.0 None
  MAP_TARGET <my_device>
  PROTOCOL WRITE LengthProtocol 32 32 8 1 BIG_ENDIAN 0 nil nil TRUE
  PROTOCOL READ LengthProtocol 32 32 8 1 BIG_ENDIAN 8 nil nil TRUE
  PROTOCOL READ_WRITE CmdResponseProtocol
```

Now I am seeing the behavior I expected! Let me know if there are any issues doing it this way.

ryanmelt, 2024-06-05:
What you are doing will work great.

It doesn't actually discard leading bytes on commands, but it will fill if you have discard and fill TRUE.

# Burst protocol
```
    def write_data(data, extra = nil)
      # If we're filling the sync pattern and discarding the leading bytes
      # during a read then we need to put them back during a write.
      # If we're discarding the bytes then by definition they can't be part
      # of the packet so we just modify the data.
      if @fill_fields && @discard_leading_bytes > 0
        data = ("\x00" * @discard_leading_bytes) << data
        if @sync_pattern
          BinaryAccessor.write(@sync_pattern, 0, @sync_pattern.length * 8, :BLOCK,
                               data, :BIG_ENDIAN, :ERROR)
        end
      end
      return super(data, extra)
    end
```

# Length protocol
```
    def write_data(data, extra = nil)
      data, extra = super(data, extra)
      if @fill_fields
        # If the start of the length field is before what we discard, then the
        # length field is outside the packet
        if @length_bit_offset < (@discard_leading_bytes * 8)
          BinaryAccessor.write(calculate_length(data.length), @length_bit_offset, @length_bit_size, :UINT,
                               data, @length_endianness, :ERROR)
        end
      end
      return data, extra
    end
```

In this case because you don't have a sync pattern, it was inserting 8 bytes of 0x00 and then filling in the length field on the commands.

closed 2024-06-05 by ryanmelt