OpenC3/cosmos #1318: Microservice UDP

Ray-13, 2024-06-05:
**Is your feature request related to a problem? Please describe.**

I can't find how to receive a send command from the UDP interface and target using a microservice.
Without the microservice, I manage to retrieve the order, but with it, I cannot find the solution. Do you have any suggestions or solutions?

**Describe the solution you'd like**

Example code for implementing this communication in a Python microservice for UDP.

**Additional context**

Currently, my Python microservice is configured to listen on a specific port, but it doesn't seem to receive the data sent by Cosmos OpenC3. Improved documentation and practical examples would greatly facilitate integration and debugging of this functionality.

```
import time
import subprocess
import sys
import socket
from openc3.microservices.microservice import Microservice
from openc3.utilities.sleeper import Sleeper

class Python(Microservice):
    def __init__(self, name):
        super().__init__(name)
        self.period = 60
        self.sleeper = Sleeper()
        self.setup_udp()

    def setup_udp(self):
        ip_address = "127.0.0.1"
        port = 8080

        try:
            self.socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
            self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
            self.socket.bind((ip_address, port))
            self.logger.info(f"Listening for UDP packets on {ip_address}:{port}")
        except OSError as e:
            self.logger.error(f"Failed to bind UDP socket: {e}")
            sys.exit(1)

    def run(self):
        self.sleeper.sleep(self.period)

        count = 0
        while not self.cancel_thread:
            start_time = time.time()

            try:
                self.socket.settimeout(10)
                self.logger.info("Waiting for UDP data...")
                data, addr = self.socket.recvfrom(1024)
                if data:
                    self.logger.info(f"Received packet from {addr}: {data}")
                else:
                    self.logger.info("Received empty packet")
            except socket.timeout:
                self.logger.info("Socket timed out, no data received.")
            except Exception as e:
                self.logger.error(f"An error occurred: {e}")

            self.state = 'RUNNING'

            run_time = time.time() - start_time
            delta = self.period - run_time
            if delta > 0:
                if self.sleeper.sleep(delta):
                    break

    def shutdown(self):
        self.socket.close()
        self.sleeper.cancel()
        self.logger.info("Closed UDP socket")
        super().shutdown()

if __name__ == "__main__":
    name = "Cosmos__SENSOR__Handler"
    Python(name).class_run()

```

```
INTERFACE <%= sensehat_target_name %>_INT udp_interface.rb host.docker.internal 8080 8081 8082 nil nil 128 nil nil

MICROSERVICE SENSORHAT sensorhat-microservice
  CMD python sensorhat.py"
```

jmthomas, 2024-06-05:
It looks like you're trying to setup a microservice to listen on a UDP port that you want COSMOS to send to. The IP `host.docker.internal` is effectively the host machine ... not the docker container where the microservice lives. Have you tried `127.0.01` or `localhost`?

Ray-13, 2024-06-05:
Yes, I tried but it's the same

Ray-13, 2024-06-05:
but when I reverse the port like this: 
`  INTERFACE <%= sensehat_target_name %>_INT udp_interface.rb localhost 8081 8080 8080 none none 128 none none
`
and I send a command, the interface reacts and crashes and returns: 
` Connection refused - recvfrom(2)`

Ray-13, 2024-06-05:
It works with localhost I forgot to change it in my microservice, thank for your help

closed 2024-06-06 by Ray-13