OpenC3/cosmos #1319: Unable to build 5.16.2

Lakshmi5785-dev, 2024-06-05:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

Next search the issues here on Github. If you come up empty please submit an issue.
[image]

jmthomas, 2024-06-06:
It obviously builds for us and it's hard for me to think this is a Windows issue since we're pulling in node packages. I couldn't come up with anything obvious during a search either. I see the error is on buildLocale ... weird question but are you building on an English locale computer? Also is this a new error on a new machine ... I assume you've been able to build in the past?

Lakshmi5785-dev, 2024-06-07:
Hi Thomas,
      Yes I have build all the previous verstion and they are running fine on my machine(atleast upto 5.15 I have tried). I have trouble only with this build. And also I tried on 2 machines with same configuration and both of them got stuck with the same error. Its a local windows machine.

jmthomas, 2024-07-13:
@Lakshmi5785-dev Does this still occur on 5.16? 5.17? Are you in a batch shell or powershell? Is this OBE?

Lakshmi5785-dev, 2024-07-15:
I didn't try on 5.17 Thomos, Will give a try and inform you here this week.

jmthomas, 2024-08-29:
I think this is OBE

closed 2024-08-29 by jmthomas