OpenC3/cosmos #1323: Script runner and logs

Ray-13, 2024-06-07:
I need to write to a file using Python with a command sender.
```
from openc3.script.suite import Suite, Group
import sys
import subprocess
import time

class Power(Group):
    def script_power_on(self):
        Group.print("Verifying requirement SR-1")
        with open("logs.txt", "w") as f:  
          f.write("script_power_on") 
```
 The 'logs.txt' file is in the same folder, but it's not writing. How can I do this?

jmthomas, 2024-06-07:
If you're trying to write out a file to the bucket system see this example: https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST2/procedures/target_file.py

Ray-13, 2024-06-11:
> If you're trying to write out a file to the bucket system see this example: https://github.com/OpenC3/cosmos/blob/main/openc3-cosmos-init/plugins/packages/openc3-cosmos-demo/targets/INST2/procedures/target_file.py

closed 2024-06-11 by Ray-13