OpenC3/cosmos #1326: Sometimes not getting response from "cmd" command located on openc3/script/commands.py

MightyArty, 2024-06-10:
**Bug Description**
Hey, I usually use the `cmd` command to send commands via cosmos and there were no problems.
I've noticed that sometimes some of the commands are not returning anything, and are stuck on the `cmd` command located in openc3/script/commands.py.
Usually, if there is some exception or error raised from your side, I'm catching it and dealing with it, but here the case is that it does not return anything and just stuck on this command (I've already verified the command parameters, and manually the same command is passing).
It's very strange, because there are no specific COSMOS commands on which it failing, it could return a response for command `a' for example, and after a couple of seconds for the same command `a` it will no return a response.

**Expected behavior**
As usual, I'm expecting the command to return the response from the cosmos, so I could proceed.

**Environment (please complete the following information):**

- OS: Windows 11
- OpenC3 COSMOS Version 5.16.2

jmthomas, 2024-06-11:
Hmm, I wonder if this is related to #1316. Note that `cmd` can take a timeout parameter to change the default 5s timeout.

MightyArty, 2024-06-16:
@jmthomas Regarding the timeout parameter, I know.
I understand that the timeout parameter can be unpredictable when using the `cmd` command. There are times when I don't receive any feedback, and I'm not sure how to handle this issue. It seems like the problem is difficult to pinpoint. For example, I can send the same command 1000 times and get a response every time, but on the 1001st try, it may fail.

ryanmelt, 2024-06-16:
After it breaks is it broken forever until a restart?
If so we found a bug on Friday that might explain that.  https://github.com/OpenC3/cosmos/pull/1337/files

jmthomas, 2024-07-13:
closed by #1381

closed 2024-07-13 by jmthomas