OpenC3/cosmos #1327: Plugin installation fails if minimum OpenC3 version, specified in plugin, is not met

321github123, 2024-06-10:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We would like users of a plugin to know what versions of OpenC3 the plugin was tested against.

**Describe the solution you'd like**
Plugin installation fails if OpenC3 minimum version, specified in plugin, is not met.

**Additional context**
Add any other context or screenshots about the feature request here.

JoshuaNeely, 2024-06-10:
It would probably make sense to allow a maximum version also.
Newer COSMOS updates might break some legacy plugins at some point. It'd be nice to communicate that in the plugin if we couldn't fix it properly

jmthomas, 2024-06-11:
Maximum is a little difficult because you never know what we're going to change. :-) We only make API changes in minor releases (5.x) and try to make them backwards compatible if at all possible. COSMOS 6.x will probably remove some deprecated APIs so if your tested version is COSMOS 5.16.2 that should imply that 6.0.0 may or may not work.

ryanmelt, 2026-10-10:
We warn if version is not met, but still allow.

closed 2026-10-10 by ryanmelt