OpenC3/cosmos #1334: Multiline not workingi n combo_box api

Lakshmi5785-dev, 2024-06-14:
combo_box("Confirm if the antenna canister has deployed and the antenna has risen from the cavity. If the canister is still stowed, escape or retry.                            \n\na> ENTER to proceed to heat filars \nb> ESCAPE to abort prior to filar heating and exit the script. \nc> RETRY to wait 10 minutes and deploy again.",                            'ENTER','ESCAPE','RETRY')

When I use the above method I got the below error.

[image]

jmthomas, 2024-06-15:
Works for me:
[image]

Lakshmi5785-dev, 2024-06-15:
Ohh Yeah, when I openend new file and pasted above lines yes it worked.
When I used in my script it fail. will debug from my script perspective. Thank you for the reply.

jmthomas, 2024-06-15:
Ok I'll close but post any additional info as you find out.

closed 2024-06-15 by jmthomas