OpenC3/cosmos #1351: Unable to send cmd over CAN interface

Shivaraj199, 2024-06-18:
I am running COSMOS on linux where I am trying to communicate with external  HW  using custom protocol over CAN interface.I've followed the [Getting Started](https://docs.openc3.com/docs/v5/gettingstarted) section, creating my own plugin with my target definition . Before implementing the custom protocol I am trying to use linux socketcan with COSMOS where i can send  data frames with up to 8 bytes as command. For this i have created a file under cosmos-project/openc3-cosmos-test-can/targets/TEST_CAN/lib/can_interface.rb directory.

When I try to send the command using command Sender I get the error
"Interface not connected" . I am not sure if i am missing some configurations
I am attaching pluging.txt fo
cat plugin.txt 
# cosmos-project/openc3-cosmos-test-can/plugin.txt
VARIABLE test_can_target_name TEST_CAN

TARGET TEST_CAN <%= test_can_target_name %>
INTERFACE <%= test_can_target_name %>_INT can_interface.rb can0
  MAP_TARGET <%= test_can_target_name %>

thank you

ryanmelt, 2024-06-19:
I don't see anything wrong with your plugin.txt, but the problem is probably with can_interface.rb.  Can you attach that?

Shivaraj199, 2024-06-19:
Hello Thank you for the reply, 
Please find the attached interface file

`class CanInterface < Cosmos::Interfaces::TcpipClientInterface
  def initialize(host, write_port, read_port, read_timeout = nil, write_timeout = nil, burst = false)
    super(host, write_port, read_port, read_timeout, write_timeout, burst)
  end

  def connect
    super
    # Initi
    `sudo ip link set up can0  type can bitrate 1000000`
  end

  def disconnect
    # Bring down CAN interface
    `sudo ip link set down can0 `
    super
  end

  def write_packet(data, _target_name, _packet_name)
    `cansend can0 #{data.unpack1('H*')}`
  end

  def read_packet
    response = `candump -n 1 can0`
    return nil if response.empty?

    data = [response.split(' ').last].pack('H*')
    data
  end
end`

ryanmelt, 2024-06-19:
I've never done CAN stuff, but the first step is going to be being able to access the can bus from inside Docker.   
See: https://stackoverflow.com/questions/63019226/sharing-virtual-network-with-docker-container

Shivaraj199, 2024-06-20:
Hello Thank you for the information, 

So i created a container that access the CAN, via docker run --rm -it --network host --privileged --device /dev/can0  command.
When i try to ship this within the compose.yaml file, the docker is not able to  access the CAN bus

Shivaraj199, 2024-06-24:
Hello so after playing around a bit i was able to mange to access the can bus from inside Docker. I verified by the starting the container.
Now when i try to run the GUI I see the error 

Unable to require can_interface.rb due to uninitialized constant OpenC3::Interfaces::Interface. Ensure can_interface.rb is in the OpenC3 lib directory.

Thanks in advance

ryanmelt, 2024-06-24:
Can you please share what you did to allow access to the CAN bus inside Docker?  That will help our other users.

For you interface error:

Add:

require 'openc3/interface/tcpip_client_interface.rb'

# This line should not have ::Interfaces
class CanInterface < Openc3::TcpipClientInterface

Shivaraj199, 2024-06-25:
Hello Thank you for the not  inteface is connected but when i try to send CMD from cmd sender i get interface not writable error

Yes ofcourse,  i actually just followed the steps from the link you shared.
1. First need to install can  utils lib in the conatiner. i used cosmos-openc3-cosmos-script-runner-api.
2. Download the vxcan kernel modules on the HOST SYSTEM 
3.Step up the vxcan and pass it too the container using sudo nsenter -t $DOCKERPID -n ip link set vxcan1 up
4. you can inspect the container from docker exec -it $CONTAINER_ID command.
5. Manually write a Script that uses vxcan and can execute the script script runnner

closed 2024-07-02 by Shivaraj199