OpenC3/cosmos #1353: Raise warning with mismatched ID_ITEM types

micromashor, 2024-06-20:
**Describe the bug**
If the ID_ITEMs are inconsistent across telemetry packets (in my case, INT vs UINT), no warning is raised, but only the first datatype will be followed. This can lead to some frustrating troubleshooting.

**To Reproduce**
Steps to reproduce the behavior:

```
TELEMETRY TARGET PACKET_A LITTLE_ENDIAN
  ID_ITEM COMMAND_ID 0 8 INT 100
  ...

TELEMETRY TARGET PACKET_B LITTLE_ENDIAN
  ID_ITEM COMMAND_ID 0 8 UINT 200
  ...
```

In this case, no packet will ever match PACKET_B, because the field is treated as a signed integer, so values >127 are treated as negative. Therefore, no packet will match the value 200. After changing PACKET_A to UINT, all packets matched correctly and worked as expected.

It is the user's fault for defining the packets differently, but it would save some headache if this triggered a warning.

**Expected behavior**
Raise a warning when installing a plugin that is defined like this, e.g.

WARNING: ID_ITEMs for bits 0..7 differ between PACKET_A (INT) and PACKET_B (UINT)

**Environment (please complete the following information):**

- OS: RHEL 8.9
- OpenC3 COSMOS Version: 5.16.2
- Browser: Chrome 125 Stable 64-bit on Win 11

ryanmelt, 2024-06-25:
related to #1168

jmthomas, 2026-01-01:
This is now handled automatically as we detect the unique IDs

closed 2026-01-01 by jmthomas