OpenC3/cosmos #1355: Sometimes unable to see the tabs in OpenC3

Lakshmi5785-dev, 2024-06-23:
Sometimes the container looks fine, but I didn't see anyother tabs apart from admin in the OpenC3

[image]
[image]

ryanmelt, 2024-06-23:
If the init container is still running then all the tools aren't installed yet.

Lakshmi5785-dev, 2024-06-23:
So should I just wait?Ideally it will work immediately after building but its been 15mnts .
Below are the logs, If needed can attach full logs.

2024-06-23 13:23:03 Restored openc3-cosmos-tool-admin-5.13.1.pre.beta0.20240623145643
2024-06-23 13:23:04 Restoring gems to pristine condition...
2024-06-23 13:23:04 Restored openc3-cosmos-tool-iframe-5.13.1.pre.beta0.20240623145741
2024-06-23 13:23:04 Restoring gems to pristine condition...
2024-06-23 13:23:04 Restored openc3-tool-base-5.13.1.pre.beta0.20240623145544
2024-06-23 13:23:04 Sun Jun 23 17:23:04 UTC 2024
2024-06-23 13:23:04 2024-06-23T17:23:04Z waiting for Minio http://openc3-minio:9000 RC: 0
2024-06-23 13:23:05 2024-06-23T17:23:05Z waiting for Redis. RC: 0
2024-06-23 13:23:06 2024-06-23T17:23:06Z waiting for Redis Ephemeral. RC: 0
2024-06-23 13:23:07 ruby: warning: shebang line ending with \r may cause problems
2024-06-23 13:23:08 Checking for needed migrations...
2024-06-23 13:23:08 Migrations complete
2024-06-23 13:23:08 ruby: warning: shebang line ending with \r may cause problems
2024-06-23 13:23:09 Added `openc3minio` successfully.
2024-06-23 13:23:09 Created policy `script` successfully.
2024-06-23 13:23:10 Added user `scriptrunnerminio` successfully.
2024-06-23 13:23:10 mc: <ERROR> Unable to make user/group policy association. The specified policy change is already in effect. (Specified policy update has no net effect).
2024-06-23 13:23:10 ruby: warning: shebang line ending with \r may cause problems
2024-06-23 13:23:11 ruby: warning: shebang line ending with \r may cause problems
2024-06-23 13:23:12 No version change detected for: openc3-tool-base-5.13.1.pre.beta0.20240623145544.gem__20240623171042
2024-06-23 13:23:12 ruby: warning: shebang line ending with \r may cause problems
2024-06-23 13:23:13 No version change detected for: openc3-cosmos-tool-iframe-5.13.1.pre.beta0.20240623145741.gem__20240623171054
2024-06-23 13:23:13 ruby: warning: shebang line ending with \r may cause problems
2024-06-23 13:23:13 No version change detected for: openc3-cosmos-tool-admin-5.13.1.pre.beta0.20240623145643.gem__20240623171057
2024-06-23 13:23:13 ruby: warning: shebang line ending with \r may cause problems

ryanmelt, 2024-06-23:
Shouldn't take 15 minutes unless you have a very slow computer.

Lakshmi5785-dev, 2024-06-23:
Yes, I don't know once in a while this happens. Trying to debug.

Lakshmi5785-dev, 2024-06-23:
RuntimeError : DEFAULT__openc3_targets:<TARGET_NAME> already exists at create
I could see this error in my init container logs. What does this mean?

Lakshmi5785-dev, 2024-06-23:
[image]
I changed my docker data location to d drive instead of c . will this cause an issue?

Lakshmi5785-dev, 2024-06-23:
Error: Unknown keyword and parameters for Target: MAP_TARGET <TARGET_NAME>  -> What does this mean?

Lakshmi5785-dev, 2024-06-23:
well I tried gem cleanup. Then it worked fine.

closed 2024-06-23 by Lakshmi5785-dev