OpenC3/cosmos #1361: Issue with running scripts 5.16.2

bradenvirdellgenevatech, 2024-06-25:
Hello,

I have been attempting to install and run the 5.16.2 version of Cosmos enterprise. Everything seems to go fine until I attempt to run scripts. When running in a as a local only environment as in its set to (localhost:2900) the scripts are running fine but when I attempt to set  it up to use SSL can no longer run scripts and I receive a 500 error with no other information.   
Here are the steps followed.
1. Pulled Openc3-enterprise from github
2.  Updated .env file with the following:
     a. OPENC3_TAG=5.16.2
     b. OPENC3_ENTERPRISE_TAG=5.16.2
     c. Commented out OPENC3_LOCAL_MODE=1
     d. Commented out OPENC3_DEMO=1
3. Changed the following in the compose.yaml
     a.  In openc3-traefik
          - Commented out "127.0.0.1:2900:2900" in ports
          - Commented out "127.0.0.1:2943:2943" in ports
          - Set "80:2900" -in ports
          - Set "443:2943" in ports
     b. In openc3-keycloak
          -  Set KC_HOSTNAME to my desired IP 
          -  Set KC_HOSTNAME_PORT to "-1"
4. Created and place the cert.crt and cert.key in the openc3-enterprise-traefik folder
5. ran ./openc3.sh run
6. Went to the keycloak webpage and set the "Valid redirect URIs" to "https://<our ip>/*"
7. I was then able to login to the main COSMOS page and upload gems
8. When attempting to run a script it would then hit a 500 error.

To me it seems I have probably missed a step somewhere which is causing this internal error but I'm not sure what that could be, I could also see it being some sort of permission errors. Any help would be appreciated. I can give more detail if needed.

ryanmelt, 2024-06-25:
Running scripts require the offline_access role in Keycloak.  Does your user have that?

bradenvirdellgenevatech, 2024-06-25:
I have tried it with 2 user both have the offline_access role in role mappings
[image]

ryanmelt, 2024-06-25:
Can you look at the docker logs for the script-runner-api container and see more info on the error.  I would expect a stack trace

bradenvirdellgenevatech, 2024-06-25:
Looks like the offline access token is invalid 
[image]

ryanmelt, 2024-06-25:
Please setup a support call for tomorrow and we can debug:
https://calendly.com/openc3/openc3-30-minute-support-call?back=1&month=2024-06

I think the issue is probably the Keycloak settings given to the Script-runner-api container but would be easier to debug if I can see those details.

bradenvirdellgenevatech, 2024-06-25:
Hey Ryan
The link you sent doesn't show any times available for tomorrow. Is there
another preferred way to schedule this?

ryanmelt, 2024-06-25:
I opened some available for tomorrow.  Please try again.

bradenvirdellgenevatech, 2024-06-26:
Quick update, moved keycloak back to the version used in 5.15.2 and things are working properly. So the assumption that its a keycloak update seems to be correct.

jmthomas, 2024-07-13:
We went from keycloak 22 to keycloak 24 from 5.15.2 to 5.16.1 (https://github.com/OpenC3/cosmos-enterprise/compare/v5.15.2...v5.16.1). This was maybe a mistake to jump 2 major releases. I don't see anything major in keycloak 23 (https://www.keycloak.org/docs/23.0.0/upgrading/#migration-changes) but keycloak 24 did introduce a lot of new changes (keycloak 23 (https://www.keycloak.org/docs/24.0.0/upgrading/#migration-changes). We've never seen backwards incompatibility though so not sure where you guys got during the debugging but we should try to track this down.

ryanmelt, 2024-07-13:
This is a very specific use case of using SSL, with an IP address rather than a hostname, and a self-signed certificate.   Newer versions of Keycloak seem to have become more restrictive to not like that scenario.   

I wrote ticket #1370 to capture what I think the long term solution is (which is to properly setup a local certificate authority for the self-signed certificates using mkcert).

jmthomas, 2024-09-27:
closing in favor of #1370

closed 2024-09-27 by jmthomas