OpenC3/cosmos #1362: Custom Python interfaces seem to have connect() method called twice

heyschultzy, 2024-06-25:
I'm developing my own custom Python ActiveMQ interface based off of Interface class (doesn't seem to be one in your examples, other than using the MQTT interface to AMQ), but I noticed it appears to be calling my connect() method twice within about 3ms. That was with version 5.15.2, so I downloaded the latest 5.16.2 and created a *highly* useful interface called nada_interface.py that just has some stubbed-out methods for connecting, reading, etc. I noticed it seems to do the same thing:

[image]

My plugin.txt looks like this for my target:

```
VARIABLE mytarget_target_name MYTARGET

TARGET MYTARGET <%= mytarget_target_name %>
INTERFACE <%= mytarget_target_name %>_INT nada_interface.py host.docker.internal 61613 61613 10.0 None
  MAP_TARGET <%= mytarget_target_name %>
```

You should be able to reproduce by stubbing out your own version of a Python interface class, or I can upload mine if you need it.

This seems to be related to this issue, but doesn't appear to be fixed for my use case:

Fix python tpcip client connection,  #1034

I'm running this on Ubuntu 22.04 in Chrome.

Thanks! ~Jeff

heyschultzy, 2024-06-25:
Also, don't know if this is related or not, but when I click the Interface Disconnect button in the GUI, I see it log that the disconnect is requested, but when I then click the Connect button, I see that it logs a "Connect requested" but my connect() method is never called in the interface (since I'm logging that inside my interface), and it just hangs there in "Attempting" status.

ryanmelt, 2024-06-25:
Can you post the entire nada_interface.py?

heyschultzy, 2024-06-26:
Sure! Here it is in all its glory...
[Uploading nada_interface.py.txt…]()

heyschultzy, 2024-06-26:
Sure! Here it is in all its glory...

[nada_interface.zip](https://github.com/user-attachments/files/15989620/nada_interface.zip)

ryanmelt, 2024-06-26:
I think your issue is read_interface should never return empty data ''b.  That indicates a broken interface

ryanmelt, 2024-06-26:
Also it needs to block, or you will max out the cpu with constant data

heyschultzy, 2024-06-27:
Ok, yeah I put a sleep in there so it won't hammer the cpu, but I could make it just wait forever.

The connect() is being called twice in quick succession *before* any call to read_interface(), so I think there is still an underlying issue, no?

ryanmelt, 2024-06-27:
I still need to look into the connect twice thing.  For now I would recommend writing your code to handle that, and just do nothing on the second connect call.

heyschultzy, 2024-06-27:
Yup! That's what I did...

PR #1400 (2024-07-13): Remove extra connect in python_microservice
closes #1362
closed 2024-07-13 by jmthomas