OpenC3/cosmos #1363: Cosmos Load Error

ireger, 2024-06-25:
Hello,

This may be a question out of my non-familiarity with Ruby and Docker. But on the prior version of Cosmos without docker, I was able to add in the 'spreadsheets' Gem into the gemfile to get it to work in ruby. 
Require 'spreadsheet' code line would work in the actual ruby script file
Now I am trying to do the same thing, but I am not sure what file to edit to get different gemfiles to work or what the process is.
The error I am getting now is:
"LoadError : only relative TARGET files are allowed -- spreadsheet","type":"log"}

ryanmelt, 2024-06-25:
You can add other libraries into COSMOS using the Admin -> Packages Tab

ireger, 2024-06-26:
Wow alot easier than what I was thinking. Thanks for helping a newbie out.

closed 2024-06-26 by ireger