OpenC3/cosmos #1364: Help/documentation for custom tools / API

heyschultzy, 2024-06-25:
We are wanting to write our own custom vue/js tool with a few buttons and a plot display. We can get the buttons to work and the plot display (sigplot) comes up, but it's not obvious from the documentation on how we get access to telemetry data from an API perspective. Is there documentation on that somewhere or some example code or one of OpenC3's own tools that can be used for a reference on how to do that? Thanks!

ryanmelt, 2024-06-25:
Our own tools are the best examples.   You can find all of them here:
https://github.com/OpenC3/cosmos/tree/main/openc3-cosmos-init/plugins/packages

As far as telemetry, the web tools use the same methods available for scripting (just a javascript version)
https://docs.openc3.com/docs/guides/scripting-api#handling-telemetry

heyschultzy, 2024-06-27:
Thanks--some of those tools have a lot of code (and extra functionality) to follow. Do you have one that you would recommend as the most straightforward example?

ryanmelt, 2024-06-27:
PacketViewer is probably the simplest for polling telemetry.

jmthomas, 2025-12-11:
We should add a documentation page for the Javascript APIs