OpenC3/cosmos #1366: How to turn on debug logging

heyschultzy, 2024-06-26:
Is there any way to get debug level logs to print from Python code besides directly setting the level in my Python code? I looked around a bit in the docs and code but couldn't find anything. Thanks...

ryanmelt, 2024-06-26:
The only way to currently change log levels is in the code like you mentioned.   
This feature will be coming with the microservice control PR when completed.

ryanmelt, 2025-11-10:
This still needs to be implemented.    It should be a new pencil icon on the microservices tab in Admin, that allows changing environment variables.   Use an environment variable called OPENC3_LOG_LEVEL to set the log level and then the microservice base class should use that if defined to initialize the log level for the microservice.

jmthomas, 2025-11-12:
duplicate of #155