OpenC3/cosmos #1372: Query regarding script runner log.

Lakshmi5785-dev, 2024-07-02:
My script runs for 40 minutes, but the script runner log which is generated is storing the log messages for last 10minutes only. Is there a way I can save my compleate logs?

jmthomas, 2024-07-08:
I'm not sure what you mean. The Script Runner log contains the messages from the complete script run. You can access it by selecting Script -> Execution Status and download the log. You can also access it from Bucket Explorer by navigating to `/DEFAULT/tool_logs/sr/<YYYYMMDD>`

ryanmelt, 2024-07-08:
@jmthomas Did you verify that the scriptrunner log writer isn't creating a new file every 10 minutes?

jmthomas, 2024-07-08:
Yes I let it run for 30min and only got a single log file.

ryanmelt, 2024-07-08:
@Lakshmi5785-dev Please provide an example test script that we can use to recreate.

jmthomas, 2024-08-29:
Closing due to lack of information

closed 2024-08-29 by jmthomas