OpenC3/cosmos #1373: Python microservice crashes the operator container

micromashor, 2024-07-02:
**Describe the bug**
When loading a python microservice, the operator container goes into loop, continuously restarting until the plugin is removed from the plugins directory, and the COSMOS instance is torn down and brought back up.

**To Reproduce**
Steps to reproduce the behavior:

1. Generate a blank plugin (`openc3.sh cli generate plugin example-microservice`)
2. Generate a python microservice within that plugin (`openc3.sh cli generate microservice my_usvc --python`)
3. Build the plugin and load it into the COSMOS instance from the admin console

Also, while attempting to diagnose by running a simple python script within the container, I noticed the following crashes the container right around where `time.sleep(1)` is called:
```py
import time

i = 0

while True:
    i += 1
    print(f"tick {i}")
    time.sleep(1)
```

**Expected behavior**
The plugin should load and print a message to the log periodically (every minute by default).

**Environment (please complete the following information):**

- OS: RHEL 8.9
- OpenC3 COSMOS Version 5.17.1
- Browser: Chrome 125 Stable on Win 11
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -

micromashor, 2024-07-02:
definitely something interesting going on underneath cosmos and probably docker-related.
- cat'ing the file and piping it into python (all inside the container) causes it to work as expected
- mounting the file in compose.yaml and running it inside the container works

still digging to determine root cause

ryanmelt, 2024-07-02:
Cannot recreate.
Feels like you are out of RAM potentially.  Make sure sufficient resources are available to docker
You can exec into the operator container and then run 'htop' as a good way to check resources.

micromashor, 2024-07-08:
It's looking more like a Docker bug. Still not certain what it is, but we've moved to a different instance and it's working, so I'll go ahead and close this issue.

closed 2024-07-08 by micromashor