OpenC3/cosmos #1374: template protocol response timeout

jmbell1989, 2024-07-02:
It seems that the response timeout for the template protocol doesn't work. I've tried incrementing the timeout to a larger value (20.0) and the interface still times out at 5s. Below is my plugin entry:

INTERFACE <%= target_name %>_INT tcpip_client_interface.rb host.docker.internal 60000 60000 10.0 nil TEMPLATE 0x0A 0x6F6B0A 0 nil 1 false 0 nil false 10.0

ryanmelt, 2024-07-02:
There are different timeouts within COSMOS.  
You probably also need to pass the timeout argument to your cmd()

jmbell1989, 2024-07-02:
Thanks Ryan! How do I go about doing this? Can you give an example? Here's one of my commands:
COMMAND TARGET GET_JSON_STATUS LITTLE_ENDIAN "Get Position"
	APPEND_ID_PARAMETER 	CMD_ID 				8 		UINT 		3 3 3	"COMMAND ID"
  	APPEND_PARAMETER 		CMD_TEMPLATE 		512 	       STRING 		"M408"
  	APPEND_PARAMETER 		RSP_TEMPLATE 		4096 	STRING 		"<JSON_STATUS_RESPONSE>"
  	APPEND_PARAMETER 		RSP_PACKET 			1024 	STRING 		"JSON_STATUS_RESPONSE_PKT"

ryanmelt, 2024-07-02:
https://docs.openc3.com/docs/guides/scripting-api#cmd

```cmd("TARGET GET_JSON_STATUS", timeout: 20)```

closed 2024-07-05 by jmthomas