OpenC3/cosmos #1390: Custom Python Interface: write_interface command not being called?

mkalin2, 2024-07-10:
I have a custom interface for a USB device written in python. It inherits interface.py directly and is modeled off of udp_interface.py due to not needing a separate protocol.

The connect/disconnect and read_interface functionality is working. I have my device sending packets over the interface and am receiving them fine. However, I cannot get anything to be written out of the interface.

When I try to send a command via the command sender I get the following error:
> Error sending <my_target> TEST due to RuntimeError: Timeout of 5s waiting for cmd ack

My write_interface function is very simple: 
```
    def write_interface(self, data, extra=None):
        Logger.warn('trying to write')
        self.write_interface_base(data, extra)
        self.device.write(data)
        Logger.warn('I wrote!')
        return (data, extra)
```

I never see these Logger.warn() messages appear when I try to send a command. It seems as though write_interface is never being called. Am I missing something? Do I need to implement any other functions? How can I troubleshoot this error message? There are no other errors that show up in the CmdTlm log when using this interface...

ryanmelt, 2024-07-10:
write_interface is called by the base class Interface#write method.  That gets called from the interface_microservice.py.
Are you sure that something isn't crashing?  Take a look in the Docker logs for the operator container and look for any stack traces.

jmthomas, 2024-08-29:
Closing due to lack of information

closed 2024-08-29 by jmthomas