OpenC3/cosmos #1393: TCP Client Interface receiving muliple packet in one burst

avan989, 2024-07-11:
I am use the tcpip_client_interface.rb, and every time it reads "tlm", it reads multiple packets when I use the read_data in my custom PROTOCOL. How do I break the raw data into various packets/is there a way to return multiple raw data?

avan989, 2024-07-11:
Wouldn’t this create a backlog, there is ~5 tlm in each “stream of data”, is there a way to speed up the call to read_data or a call to pass the raw data directly to tlm?

ryanmelt, 2024-07-11:
In the end, the system has to be able to keep up with the interface.  We have customers running at over 200 packets per second.

If you have interfaces receiving data at greater than 10Hz, try adding OPTION OPTIMIZE_THROUGHPUT 0.1 to your interfaces.

avan989, 2024-07-12:
If I could avoid the read_data, I think cosmos can keep up. I have ccsds packet wrap inside an ccsds packet, do u have a recommendation of getting the inner packet to tlm without using the read_data?

ryanmelt, 2024-07-12:
Internal COSMOS processes packets.  It is the interface (and protocols) job to inject those packets.  read_data is the right place to do strip off layers.

The alternative is to leave the layered packet, and have your packet definitions expect that.

closed 2024-10-12 by jmthomas