OpenC3/cosmos #1394: Ability to rerun selected pieces of code

321github123, 2024-07-11:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
We have scripts that consist of many steps where a step is a logical grouping of spacecraft interactions e.g. (send three commands and then wait for two telemetry values to become expected values).  We would like to allow the user to be able to selectively rerun certain steps. 

**Describe the solution you'd like**
A possible solution would be adding support for jupyter notebooks e.g. ScriptRunnerInteractive would be a new page that opens a jupyter notebook?

https://jupyter-notebook.readthedocs.io/en/latest/

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-07-11:
If you select several lines of text you can right click and then select "Execute selection". This will execute the selected lines as a separate script. Note, you have to be careful with variables and references ... it's pretty easy to select lines that rely on variables higher up in the script and that will not work.

321github123, 2024-07-11:
I can't seem to find the documentation for Test Runner. =\  Could you post a link?

ryanmelt, 2024-07-11:
The Suite Runner / Test Runner docs are here:
https://docs.openc3.com/docs/tools/script-runner#running-script-suites

I don't think we have the right-click menu "Run Selection" documented.

321github123, 2024-07-12:
If we ran an entire group, do you know if there is any guaranteed ordering of the scripts that are run?

jmthomas, 2024-07-12:
Scripts are added alphabetically and they all have to start with `script_` or `test_`. So sometimes we have customers name things like `test_01_power_on`, `test_02_ramp_rate`, `test_03_power_off`, etc.

PR #1615 (2024-10-12): Document SR right click
closes #1394
closed 2024-10-12 by ryanmelt