OpenC3/cosmos #1397: Script fails with error

Lakshmi5785-dev, 2024-07-12:
While running my procedures all of sudden I am receiving the below error.
any clue on this?

[image]

jmthomas, 2024-07-12:
Do a `docker ps` and then check the logs from the cmd-tlm-api container: `docker logs cosmos-openc3-cosmos-cmd-tlm-api-1`. You can also check the operator container: `docker logs cosmos-openc3-operator-1`. Post if you see any stack traces.

Lakshmi5785-dev, 2024-07-12:
[image]
[cmd_tlm_api_logs.txt](https://github.com/user-attachments/files/16195630/cmd_tlm_api_logs.txt)
[Uploading operator_logs.txt…]()

jmthomas, 2024-07-12:
I need the logs output, not the ps output

Lakshmi5785-dev, 2024-07-12:
Attached the test files 
[cmd_tlm_api_logs.txt](https://github.com/user-attachments/files/16195911/cmd_tlm_api_logs.txt)
[operator_logs.txt](https://github.com/user-attachments/files/16195913/operator_logs.txt)

jmthomas, 2024-07-12:
I don't see anything that would indicate a failure in the cmd-tlm-api container or operator container.

jmthomas, 2024-07-13:
I see you're using Python but I can not reproduce in the DEMO with a simple test: `cmd_no_hazardous_check(f"INST CLEAR")`

Lakshmi5785-dev, 2024-07-25:
Well this error occurs when we provide wrong command parameters. Figured out the issues. Closing the ticket.

closed 2024-07-25 by Lakshmi5785-dev

jmthomas, 2024-07-25:
@Lakshmi5785-dev that was a really weird error message. Was there something specific about the command parameters that could reproduce it? I'd like to at least provide a better error message.

Lakshmi5785-dev, 2024-07-25:
Let me try on INST demo target, but I am sure its because of incorrect command parameters. It happend for me couple of times. Let me reproduce and share the details here.