OpenC3/cosmos #1405: UDP Interface not working

rmslick, 2024-07-16:
Hello,
I can communicate via udp from my plugin to program on host machine. I have a sample python script trying to receive sent commands. If I try to bind to the port 1724, I get a socket in use error. If I simply try to read from that port or use tcpdump (with udp option) to view packets sent, no data arrives when I send commands. Any guesses?

 I have port forward listed in compose.yaml and interface defined as follows:

compose.yaml:

```
  openc3-operator:
    user: "${OPENC3_USER_ID:-1001}:${OPENC3_GROUP_ID:-1001}"
    image: "${OPENC3_REGISTRY}/${OPENC3_NAMESPACE}/openc3-operator${OPENC3_IMAGE_SUFFIX}:${OPENC3_TAG}"
    restart: "unless-stopped"
    ports:
    #  - "127.0.0.1:7779:7779" # Open port for the demo router
    - "5001:5001"
    - "1724:1724/udp"
    - "5554:5554/udp"
```

```
VARIABLE comotcp_target_name CE1
	TARGET CE1 <%= comotcp_target_name %>
	INTERFACE CE1 udp_interface.rb host.docker.internal 1724 5554 nil nil 128 1000.0 nil
	MAP_TARGET <%= comotcp_target_name %>
	PROTOCOL READ_WRITE LengthProtocol 32 16 7 1 BIG_ENDIAN 0 nil nil true
	PROTOCOL WRITE ChecksumProtocol Checksum false ERROR 48 8 BIG_ENDIAN nil nil nil nil
```

ryanmelt, 2024-07-16:
You don't need the port forward if COSMOS is only writing out to that port.   The port forward is only needed for COSMOS to receive UDP.   So remove the 1724 port forward and let me know how that goes.

rmslick, 2024-07-16:
@ryanmelt Still not receiving data. The first time I try sending a command in command sender nothing happens, subsequent attempts I get this error:

`Error sending TargetName CMDNAMEdue to RuntimeError: Interface not connected: TARGETNAME`

Here is UDP test script I run on host

```
import socket

UDP_IP = "0.0.0.0"
UDP_PORT = 1724

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.bind((UDP_IP, UDP_PORT))

print(f"UDP server up and listening on {UDP_IP}:{UDP_PORT}")

while True:
    # Receive message
    data, addr = sock.recvfrom(1024)
    print(f"Received message: {data.decode()} from {addr}")
```

(Interface definition same is above)

rmslick, 2024-07-16:
@ryanmelt Removing the port forward from compose.yaml did work, I just had a process using port 1724 I didn't notice preventing me from testing it properly. It works now, thank you

ryanmelt, 2024-07-16:
Glad to help!

closed 2024-07-16 by ryanmelt