OpenC3/cosmos #1407: Having issues using imports when creating a python test suite file

pcheng-scs, 2024-07-16:
I'm not sure what I'm doing wrong. I get a `ModuleNotFoundError: No module named 'TARGET'` when I try to import my TestGroup.py file into my TestSuite.py. I tried using load_utility, and I get a `RuntimeError: load_utility not supported outside of Script Runner`.

When I combine my test_suite, and test_group into 1 file, I still get a `ModuleNotFoundError: No module named 'TARGET'`

Am I doing something wrong? Not sure how to proceed. My end goal is to have a top level (or several top level) test suite scripts call our test groups. I do not get the  `ModuleNotFoundError: No module named 'TARGET'` when the test_suite is not defined in a file.

I am using version 5.16.2.

Here's the folder structure I'm trying to use.

* lib
  * target.py
* procedures
  * test_suite.py 
  * component
    * test_group .py 	

Here are the file contents:

test_suite.py (using imports)
```python
from openc3.script.suite import Suite
from TARGET.component.test_group import TestGroup

class CompTestSuite(Suite):
    def __init__(self):
        super().__init__()
        self.add_group_setup(TestGroup)
        self.add_script(TestGroup, 'test_1')
        self.add_group_teardown(TestGroup)
```

test_suite.py (using load_utility)
```python
from openc3.script.suite import Suite

class CompTestSuite(Suite):
    def __init__(self):
        super().__init__()
        load_utility("TARGET/procedures/component/test_group.py")
        self.add_group_setup(TestGroup)
        self.add_script(TestGroup, 'test_1')
        self.add_group_teardown(TestGroup)
```
		
test_group .py
```python
from enum import Enum
from openc3.script import *
from openc3.script.suite import Group
from TARGET.lib.target import Target

SIZE_BASE_PKT = 12
CMD_ID = 10

class TestGroup(Group):
    
    def __init__(self):
       self.helper = Target("COMP", CMD_ID)

    #***************************************************************************
    # SETUP
    #***************************************************************************
    def setup(self):
      pass

    #***************************************************************************
    # Test 1
    #***************************************************************************
    def test_1(self):
      cmd(self.helper.build_cmd(SIZE_BASE_PKT, "FUNC NAME", "FUNC_CODE"))

    #***************************************************************************
    # TEARDOWN
    #***************************************************************************
    def teardown(self):
       pass
```

target.py
```python
class Target():
    def __init__(self, comp_name, cmd_id):
        self.m_comp_name = comp_name
        self.m_cmd_id = cmd_id

    def build_cmd(self, pkt_len, func_name, func_code, extra_params = None):
        cmd_params = []
		
        if not extra_params is None:
            cmd_params.extend(extra_params)

        return f'TARGET {self.m_comp_name}_{func_name} with ' + ', '.join(cmd_params)
```
		

Combined TestSuite and TestGroup.py
```python
from enum import Enum
from openc3.script import *
from openc3.script.suite import Group, Suite
from TARGET.lib.target import Target

SIZE_BASE_PKT = 12
CMD_ID = 10

class test_group (Group):
    
    def __init__(self):
       self.helper = Target("COMP", CMD_ID)

    #***************************************************************************
    # SETUP
    #***************************************************************************
    def setup(self):
      pass

    #***************************************************************************
    # Test 1
    #***************************************************************************
    def test_1(self):
      cmd(self.helper.build_cmd(SIZE_BASE_PKT, "FUNC NAME", "FUNC_CODE"))

    #***************************************************************************
    # TEARDOWN
    #***************************************************************************
    def teardown(self):
       pass

class CompTestSuite(Suite):
    def __init__(self):
        super().__init__()
        self.add_group_setup(TestGroup)
        self.add_script(TestGroup, 'test_1')
        self.add_group_teardown(TestGroup)

```

GabeBarfield, 2024-10-23:
We are having the same issue as well using the COSMOS repo version 5.17.1.

Folder structure is
-TGT
   -procedures
      -file1.py

The import line looks as follows:
**from TGT.procedures.file1 import ***
This throws the error: **ModuleNotFoundError: No module named 'TGT'**

However we can use load_utility('TGT/procedures/file1.py) fine but we prefer not to instrument this file.

ryanmelt, 2024-10-28:
This is working for me just fine at least on our working main branch.

```
from INST2.procedures.my_test_suite import *

a = MyTestSuite()
```

Please try our latest release and let us know if it is not working for you.

closed 2024-10-28 by ryanmelt