OpenC3/cosmos #1408: Autonomic API

bradenvirdellgenevatech, 2024-07-16:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
Currently we use quite a few custom microservices, we want to reduce this number and are looking to leverage autonomic to do so. The issue  is that you have to create all triggers and reactions manually. We would like a simpler way to source control autonomic triggers and reactions. Thus would like a way to upload a file with autonomic configurations. 

**Describe the solution you'd like**
We believe an API that can create Autonomic triggers and reactions like the Calendar API, would work well as it would have a lot of flexibility and other uses.

jmthomas, 2024-07-19:
This is clearly a hole in our APIs ... good suggestion.

thomasstjeangenevatech, 2025-02-25:
@jmthomas Any update on this? Is there a plan to implement this in a future release?

jmthomas, 2025-02-25:
I'll make this a priority. I think we can get to this for the 6.3.0 release sometime in March.

closed 2025-05-29 by jmthomas