OpenC3/cosmos #1409: Data Extractor file name times not picking up current time

sunws4me, 2024-07-16:
time does not change on multiple runs of data extractor over different system clock times.

setup a data extraction and repeated running over multiple seconds, however, the filename on each output was the same - the seconds field did not increment to match the actual run time

clicking into packet viewer and back to data extractor and rerunning does get a new timestamp as part of the new file name.
[image]

ryanmelt, 2024-07-16:
I think this is as expected.  The filename is based on the data extraction start time field.   (Not current system clock time)

sunws4me, 2024-07-16:
Hello,
Just thought it was odd to have a static file name on the output, current system time is a nice incrementing value and then the user gets an easy history without doing any work to sort it out and rename files.

ryanmelt, 2024-07-16:
We could potentially have different file naming modes, but I like how it currently works.   It relates to what you are extracting, rather than whatever time it happens to be right now.

sunws4me, 2024-07-16:
A little confused by that,  ‘relates to what you are extracting’.  The file name stays static from when you went to the page and doesn’t change, that’s not going to relate to anything the user is doing its just a random point in time, not where they were on 9/11…..  system time as part of the output name wouldn’t relate either, just a name that would be changing over time and not include the dup markings (1), (2), …

I haven’t done the save/load, if I load a saved config, will the name of the config be part of the output name?  that would be a piece of meaningful data if its part of the filename (mine would be bob so maybe not for me!)

Since its saving to a file its really not a big deal, everything I’ve done in my history has always included system time at least so I thought it was a bug.

ryanmelt, 2024-07-16:
Its not a random point in time.  Its the time in the start field.

ryanmelt, 2024-07-16:
[image]

sunws4me, 2024-07-17:
Thanks I was missing that piece.

closed 2024-07-20 by ryanmelt