OpenC3/cosmos #1417: Keycloak/WebSocketApi Authentication Issues

stephen-ritter, 2024-07-18:
### Issue Statement

There appear to be two somewhat related issues I have stumbled upon when using the WebSocketApi.
1. The use of the environment variable `OPENC3_API_TOKEN` instead of `OPENC3_API_PASSWORD` seems incorrect.
    - This can be bypassed by setting `OPENC3_API_TOKEN` to anything to make it go down into the Keycloak authentication logic as a solution to get it to work.
    - https://github.com/OpenC3/cosmos/blob/1ee0440edd2c7e35cf3ae77888ca26a05750ce3c/openc3/python/openc3/script/web_socket_api.py#L152 
2. The string `Bearer ` being attached to the token seems incorrect and this causes invalid URLs when using the token.
    - https://github.com/OpenC3/cosmos/blob/1ee0440edd2c7e35cf3ae77888ca26a05750ce3c/openc3/python/openc3/utilities/authentication.py#L82
    - https://github.com/OpenC3/cosmos/blob/1ee0440edd2c7e35cf3ae77888ca26a05750ce3c/openc3/python/openc3/script/web_socket_api.py#L129

### Hypothetical Solutions
- Note: The proposed solutions have not been fully tested and I am unsure if these fixes need to be applied to the Ruby side of the code as well as I have only been testing the Python side.
1. Change `OPENC3_API_TOKEN` to `OPENC3_API_PASSWORD`
2. Have `token()` just `return self._token` instead of `return f"Bearer {self._token}"`

### Demo Script
This demo script is basically a copy and paste of the `cosmos_web_socket_example.py` script with some of the environment variables changed to match my system. I did leave `os.environ["OPENC3_API_TOKEN"] = "THIS_IS_AN_ERROR"` commented out, but to make this work that lines needs to be uncommented. I am just demonstrating the error.
```python
import os
from datetime import datetime, timezone, timedelta

os.environ["OPENC3_API_USER"] = "operator"
os.environ["OPENC3_API_PASSWORD"] = "operator"
# This needs to be set to anything as the WebSocketApi is checking to see if this environment variable is set
# If it remains unset the following error is thrown: RuntimeError: Unauthorized
# os.environ["OPENC3_API_TOKEN"] = "THIS_IS_AN_ERROR"
os.environ["OPENC3_KEYCLOAK_URL"] = "http://localhost:2900/auth"
os.environ["OPENC3_SCOPE"] = "DEFAULT"
os.environ["OPENC3_API_HOSTNAME"] = "localhost"
os.environ["OPENC3_API_PORT"] = "2900"
os.environ["OPENC3_SCRIPT_API_HOSTNAME"] = "localhost"
os.environ["OPENC3_SCRIPT_API_PORT"] = "2900"

from openc3.utilities.time import to_nsec_from_epoch
from openc3.script.web_socket_api import StreamingWebSocketApi, MessagesWebSocketApi

with StreamingWebSocketApi() as api:
    api.add(
        # Get a list of individual telemetry items
        items=[
            "DECOM__TLM__INST__HEALTH_STATUS__TEMP1__CONVERTED",
            "DECOM__TLM__INST__HEALTH_STATUS__TEMP2__CONVERTED",
        ]
        # You can alternatively request entire packets
        # packets=[
        #     "DECOM__TLM__INST__HEALTH_STATUS",
        #     "RAW__TLM__INST__HEALTH_STATUS",
        # ]
    )
    for _ in range(5):
        print(api.read())
    api.remove(items=["DECOM__TLM__INST__HEALTH_STATUS__TEMP1__CONVERTED"])
    for _ in range(5):
        print(api.read())

# Warning this saves all data to RAM. Do not use for large queries
now = datetime.now(timezone.utc)
data = StreamingWebSocketApi.read_all(
    items=[
        "DECOM__TLM__INST__HEALTH_STATUS__TEMP1__CONVERTED",
        "DECOM__TLM__INST__HEALTH_STATUS__TEMP2__CONVERTED",
    ],
    start_time=now - timedelta(seconds=30),
    end_time=now + timedelta(seconds=5),
)
print(data)

now = datetime.now(timezone.utc)
with MessagesWebSocketApi(
    history_count=0,
    start_time=to_nsec_from_epoch(now - timedelta(seconds=3600)),
    end_time=to_nsec_from_epoch(now - timedelta(seconds=60)),
) as api:
    for _ in range(500):
        # Note returns batch array
        data = api.read()
        if not data or len(data) == 0:
            break
        print(f"\nReceived {len(data)} log messages:")
        print(data)
```

stephen-ritter, 2024-07-18:
I just realized for my first point that the code has been changed. I have been testing on 5.15.2 but it looks like `is None` has been added to those environment variable checks, which most likely fixes the issue with point 1.

jmthomas, 2024-07-19:
There were a lot of Python fixes in the last two releases. Can you reproduce with 5.17.1?

stephen-ritter, 2024-07-19:
> There were a lot of Python fixes in the last two releases. Can you reproduce with 5.17.1?

I pulled down the latest 5.17.1, and the first point about the `OPENC3_API_TOKEN` has been fixed. I updated my testing script to remove the environment variable from it to reduce confusion. The second issue still seems to be a problem. I am getting the following error:
```
2024-07-19 09:25:05,897 ERROR websockets.client unexpected internal error
Traceback (most recent call last):
  File "/home/ritters/Projects/python3.12_venv/lib/python3.12/site-packages/websockets/sync/connection.py", line 593, in recv_events
    assert not self.protocol.events_received()
AssertionError
Traceback (most recent call last):
  File "/home/ritters/Projects/test.py", line 59, in <module>
    with StreamingWebSocketApi() as api:
  File "/home/ritters/Projects/python3.12_venv/lib/python3.12/site-packages/openc3/script/web_socket_api.py", line 58, in __enter__
    self.connect()
  File "/home/ritters/Projects/python3.12_venv/lib/python3.12/site-packages/openc3/script/web_socket_api.py", line 136, in connect
    return self.stream.connect()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/ritters/Projects/python3.12_venv/lib/python3.12/site-packages/openc3/streams/web_socket_client_stream.py", line 34, in connect
    self.connection = connect(
                      ^^^^^^^^
  File "/home/ritters/Projects/python3.12_venv/lib/python3.12/site-packages/websockets/sync/client.py", line 289, in connect
    connection.handshake(
  File "/home/ritters/Projects/python3.12_venv/lib/python3.12/site-packages/websockets/sync/client.py", line 90, in handshake
    raise ConnectionError("connection closed during handshake")
ConnectionError: connection closed during handshake
```
This is caused by a malformed URL that I mentioned above. This is what the URL is without a fix to it:
`ws://localhost:2900/openc3-api/cable?scope=DEFAULT&authorization=Bearer <access token>`
Notice the `Bearer ` string (with the space) is in the URL. A dirty fix to this would be to change:
`final_url = self.url + f"?scope={self.scope}&authorization={self.authentication.token()}"`
to
`final_url = self.url + f"?scope={self.scope}&authorization={self.authentication.token()[7:]}`
This will just chop off the `Bearer ` portion of the string.

PR #1424 (2024-07-20): Fix python web socket in Enterprise
closes #1417 

This is a bit of a hack because the "right way" as described here: https://websockets.readthedocs.io/en/stable/topics/authentication.html#machine-to-machine-authentication doesn't seem to work. I get alternating results of `RuntimeError: Unauthorized` followed by `websockets.exceptions.ConnectionClosedOK: received 1000 (OK); then sent 1000 (OK)`.
closed 2024-07-20 by jmthomas