OpenC3/cosmos #1429: Add reset button to Telemetry Grapher graph

321github123, 2024-07-23:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
User would like to clear our all currently plotted telemetry items within a Telemetry Grapher graph

**Describe the solution you'd like**
Add "reset" button that clears out all previously plotted telemetry items to a Telemetry Grapher graph

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-07-24:
We have a File menu option to reset Telemetry Grapher that clears everything and closes the graph.

321github123, 2024-07-24:
Would it be possible to add a reset button that clears the graph but doesn't close it?

jmthomas, 2024-07-24:
Keep all the graph items but just clear the graph?

321github123, 2024-07-24:
Yes -- maybe the button should be called "clear"

ryanmelt, 2024-07-24:
I think there are two different capabilities mentioned here:
1. Clear all the data from a graph
2. Remove all the items from a graph?

jmthomas, 2024-07-24:
We already have the capability to clear the graph completely (remove all the items and data) via the File->Reset Configuration. I think the ability to clear all the data but keep the items is a nice feature. Not sure exactly where the right place for that option is (menu vs GUI element).

sunws4me, 2024-07-24:
yes, just remove the data from existing graph(s).  kind of need two opens, clear only 1 particular graph or all.  leave all existing points as is, just erase current line(s)

jmthomas, 2024-07-24:
Note you can already right-click an item in the legend and delete it

sunws4me, 2024-07-24:
looking for an easy way to get from this:
[image]
to this:
[image]

with just a single click

the mnemonic was hidden to protect the innocent :)

jmthomas, 2024-07-24:
Agreed, we will implement a clear data option

jmthomas, 2024-09-25:
In the PR I added the ability to right click and item and clear just its data. Or in the TlmGrapher Graph menu you have the option to clear ALL data from all items on all graphs.

PR #1570 (2024-09-25): Add ability to clear graph data
closes #1429
closed 2024-09-25 by jmthomas