OpenC3/cosmos #1430: Add persistence to Data Extractor page

321github123, 2024-07-23:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
During a long data extraction, if a user clicks away from the page and then comes back the previous data extraction is gone.

**Describe the solution you'd like**
If a user navigates away from a data extraction that is in progress, the user can return to the data extraction.

Another solution might be an individual data extraction runs in separate thread and progress bar is shown

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2024-07-24:
The items are "remembered" when navigating away but we are changing the time range each time. If you want to fully record your session you need to File->Save Configuration.

If you're asking for the page to continue an extraction in progress if you navigate away I'm afraid that's just not possible. The frontend is receiving the data streaming from the backend and closing or navigating away from the page is effectively just stopping the stream.

ryanmelt, 2024-07-24:
We would have to add a background data extractor functionality that does the work in the backend, instead of the frontend.

321github123, 2024-07-24:
The background data extractor would be a nice improvement, especially since data extractions take a while