OpenC3/cosmos #1440: API Metrics Can Grow Very Large After Running Many Unique Scripts

ryanmelt, 2024-07-26:
The Prometheus API Metrics gather we use keeps a unique metric for every API endpoint.  Our Script endpoints include the running script id, so every hit to a new url creates a new metrics.  These metrics don't appear to ever get cleaned up and can grow quite large over time with running many scripts.

We should probably build a custom DataStore object to cleanup old metrics over time to prevent large growth.
See: https://github.com/prometheus/client_ruby

ryanmelt, 2024-07-29:
Also short term have the option to disable with an environment variable.

JoshuaNeely, 2024-08-09:
Adding another symptom/side-effect that'll affect more normal workflows- even without running lots of scripts.

I have a prometheus pointed at the `/metrics` endpoint.
I plotted graph with the `cosmos_resident_memory_bytes` metric, along with the `sum`of all these entries.
The sum was growing forever!
These short lived microservices report consuming memory (and cpu, etc, I assume) even after they are long gone, making some of these graphs/metrics very misleading

[image]