OpenC3/cosmos #1447: Downsample Data

jmbell1989, 2024-08-01:
Is there a way to downsample telemtry data from an interface? I currently have periodic telemetry coming from a device at around 100hz but I only need 1hz. I know I can downsample at the telemetry extractor level but I was wondering if there was a simple method to do it at the protocol level so the extra data doesn’t even get logged.

thanks in advance!

ryanmelt, 2024-08-01:
We don't have a downsample prebuilt, but that would be a great addition.

It will be almost the same as our ignore_packet protocol: https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/protocols/ignore_packet_protocol.rb

In fact, I will probably just make it an additional optional argument to that protocol.

Until then, I recommend you create a subclass of IgnorePacketProtocol, and modify the read_packet method to not always return :STOP for the specified packet, but instead count packets and return :STOP most of the time, but let a packet through at 1Hz.

jmbell1989, 2024-08-26:
Thanks @ryanmelt! Where do I add this helper protocol? I tried adding it in the target directory but the logs say it can find the file. I can create a new protocol entirely as a subclass of say the fixed protocol and it finds it just fine. But not a helper protocol. Unless I’m calling it incorrectly in my plugin. 

And help is appreciated!

ryanmelt, 2024-08-26:
I would recommending putting it into a top level "lib" folder in your plugin.