OpenC3/cosmos #1448: Possible bug in disconnecting/re-connecting custom Python interface?

heyschultzy, 2024-08-01:
Not sure if this is a bug or some sort of issue with our implementation, but using our custom ActiveMQ interface code, we can click on Disconnect in the CmdTlmServer screen and see our disconnect() get called, but then clicking Connect doesn't seem to call our connect() method again. I *think* we're implementing our connect(), connected(), and disconnect() methods correctly, including the calls to super().connect()/disconnect().

Has this been tested before and known to work with a custom interface? I'm assuming the built-in ones like TCP work but it's been a while since I tried that.

We're running 5.16.2 on Ubuntu 22.04 with the Python API/interfaces.

ryanmelt, 2024-08-01:
Yes this works.

If your calls to connect/disconnect are getting stuck or crashing, then that will break it.

ryanmelt, 2024-08-01:
Take a look in the operator docker logs

heyschultzy, 2024-08-02:
Thanks! Will do...

heyschultzy, 2024-08-02:
@ryanmelt Tried that and nothing interesting. I am logging where my disconnect() gets called and after that I log where super().disconnect() has returned, and then that's the last thing the log reports. When I click Connect in the UI I see a "XXXXX_INT: Connect requested" but it never actually calls my interface's connect() method or __init__() or anything.

Have you tried this with the Python version of something that is a child class of Interface vs. StreamInterface?

I could test with my famous nada_interface.py next if that would help...

heyschultzy, 2024-08-02:
Is this related at all to the connect() twice bug #1362 ?

jmthomas, 2024-08-02:
I don't think it's related but you could always build from source and try against main. All of our python interfaces derive from `Interface` and they all call `super().connect()` in the connect method and `super().disconnect()` in the disconnect method. I can click Disconnect and Connect in the demo and it works. To be fair, that's a simulated interface.

heyschultzy, 2024-08-05:
Yeah, I tried with my nada_interface.py and it seems to work fine. :thinking: I still can't figure out why OpenC3 won't call my connect() method in my real plugin/target. Will continue to poke around...

heyschultzy, 2024-08-07:
Ok, this ended up being a strange one...

A little background: Because ActiveMQ normally operates by registering a callback on new messages, but OpenC3 expects to call a read_interface(), I implemented the AMQ side by having the on_message() add the topic and message to a q=queue.Queue() object. So to implement the read_interface() as blocking (I think because y'all said that would be the normal way to do it), I do a blocking call to q.get() with no timeout. Therein seems to be the problem. Disconnecting the AMQ interface seems to call my interface's disconnect(), but I guess the read_interface() is still sitting there waiting for a message in the queue. Not sure how your call to read_interface() is set up, but I'm assuming it must be a separate thread, because my disconnect() can be called while read_interface() is blocking. I saw a note in the queue [docs](https://docs.python.org/3.11/library/queue.html) saying that on older versions of Python and on Windows Queue.get() is non interruptible, but I'm not sure how that applies here. Is the OpenC3 code waiting on a thread to join/end and expecting the read_interface() to eventually exit/return on its own if the underlying interface goes down?

If I implement the underlying q.get() with a timeout inside a loop where I check for a disconnect, things seem to work now, but not sure if that is the expected approach or if I'm missing something.

jmthomas, 2024-08-08:
There's a lot of different ways to handle this. On the [Ruby MQTT interface](https://github.com/OpenC3/cosmos/blob/6540996091320518380ff516409ec9fe245b2801/openc3/lib/openc3/interfaces/mqtt_interface.rb#L4) we send a disconnect message to ourselves when disconnecting. In other places we'll wait on a queue and send an empty message to ourselves to break the connection and then cleanly disconnect. The [TcpipServerInterface](https://github.com/OpenC3/cosmos/blob/6540996091320518380ff516409ec9fe245b2801/openc3/python/openc3/interfaces/tcpip_server_interface.py#L4) uses a `threading.Condition` wrapping a `threading.RLock` to control the disconnect in the server due to the use of multiple threads.

See if any of those solutions make sense here and otherwise if you have something that works then go with it!

heyschultzy, 2024-08-09:
Ah, clever ideas! I'll give the empty/disconnect message a shot. I think that will be cleaner than doing a loop around a read with a timeout. Maybe you should add a note in your [Interfaces](https://docs.openc3.com/docs/configuration/interfaces) doc to that effect (e.g. "Methods such as sending an empty message to oneself can break the blocking read.").

Also, I took a look at the tcpip_server_interface.py but happened to see something confusing. It doesn't look like the get_nowait() will block as the comments suggest:

```
    # @return [Packet] Latest packet read from any of the connected clients.
    #   Note this method blocks until data is available.
    def read(self):
        if not self.connected:
            raise RuntimeError(f"Interface not connected for read: {self.name}")
        if not self.read_allowed:
            raise RuntimeError(f"Interface not readable: {self.name}")

        try:
            packet = self.read_queue.get_nowait()
        except queue.Empty:
            return None

        if packet is not None:
            self.read_count += 1
        return packet
```

ryanmelt, 2024-08-09:
You are correct that doesn't look right.   It appears the Python tcpip_server will hard spin on read and waste a lot of CPU.  We'll need to fix that.

heyschultzy, 2024-08-09:
Oh, haha--glad to return some favor then! :smile:

ryanmelt, 2024-08-09:
When you receive the empty message, check for it and return None, instead of a blank string.

heyschultzy, 2024-08-09:
Hmmm... the base interface doesn't seem to like that:

```
Microservice DEFAULT__INTERFACE__MYINT_INT dying from exception
['Traceback (most recent call last):
', '  File "/usr/lib/python3.11/site-packages/openc3/microservices/microservice.py", line 51, in class_run
    microservice.run()
', '  File "/openc3/python/openc3/microservices/interface_microservice.py", line 503, in run
    packet = self.interface.read()
             ^^^^^^^^^^^^^^^^^^^^^
', '  File "/usr/lib/python3.11/site-packages/openc3/interfaces/interface.py", line 162, in read
    data, extra = self.read_interface()
                  ^^^^^^^^^^^^^^^^^^^^^
', '  File "/tmp/tmptyfrhqgd/targets/MYINT/lib/amq_interface.py", line 242, in read_interface
    self.read_interface_base(data, extra)
', '  File "/usr/lib/python3.11/site-packages/openc3/interfaces/interface.py", line 398, in read_interface_base
    self.bytes_read += len(data)
                       ^^^^^^^^^
', "TypeError: object of type 'NoneType' has no len()
"]
```

Do I need to re-define read_interface_base() so it doesn't call the super() one if data is None?

ryanmelt, 2024-08-09:
Your interface should not call read_interface_base if None is going to be returned.

heyschultzy, 2024-08-09:
Ok, that seems to work. It does produce this kinda funny sounding artifact in the logs:

```
MYINT_INT: Disconnect requested
MYINT_INT: read_interface requested disconnect
MYINT_INT: Internal disconnect requested (returned None)
MYINT_INT: Connection Lost
```

Any getting around that or is that just the way it's going to be?

ryanmelt, 2024-08-09:
At least for now, that's what a healthy disconnect looks like.

jmthomas, 2024-10-12:
Python TcpipServerInterface fixed in https://github.com/OpenC3/cosmos/pull/1578

closed 2024-10-12 by jmthomas