OpenC3/cosmos #1449: Issue with Array Indices in Item Name

stephen-ritter, 2024-08-02:
There appears to be a bug that was introduced in commit f4b1a85. The problem I am seeing is that if telemetry items have array indices in their item name then the `VALUE` widget will fail to display their value because it is chopping off the index (`[n]`) and just querying the partial item name itself. I wonder if an escape character needs to be used if you want to use the functionality of the `VALUE` widget retrieving the index of an `ARRAY_ITEM`.

This is the error I see in the Telemetry Viewer:
```
1722616763720: (error) {
  "name": "RuntimeError",
  "message": "Item 'MYTARGET STATUS MYITEM' does not exist"
}
```

Here is an example telemetry packet definition and an associated screen demonstrating the problem:
```
TELEMETRY MYTARGET STATUS BIG_ENDIAN "Telemetry description"
  APPEND_ITEM    MYITEM[0] 8       UINT    "Index 0"
  APPEND_ITEM    MYITEM[1] 8       UINT    "Index 1"
  APPEND_ITEM    MYITEM[2] 8       UINT    "Index 2"
  APPEND_ITEM    MYITEM[3] 8       UINT    "Index 3"
  APPEND_ARRAY_ITEM MYARRAY         8 UINT 80 "Array data"
```

```
SCREEN AUTO AUTO 1.0

VERTICAL
  LABELVALUE MYTARGET STATUS MYITEM[0]
  LABELVALUE MYTARGET STATUS MYITEM[1]
  LABELVALUE MYTARGET STATUS MYITEM[2]
  LABELVALUE MYTARGET STATUS MYITEM[3]
END
```

ryanmelt, 2024-08-02:
You have to double bracket in screens now if your item names have array like syntax.

```
SCREEN AUTO AUTO 1.0

VERTICAL
  LABELVALUE MYTARGET STATUS MYITEM[[0]]
  LABELVALUE MYTARGET STATUS MYITEM[[1]]
  LABELVALUE MYTARGET STATUS MYITEM[[2]]
  LABELVALUE MYTARGET STATUS MYITEM[[3]]
END
```

Give that a shot

stephen-ritter, 2024-08-02:
That worked, but I see some issues with this though.

- This is not backwards compatible with previous versions of Cosmos which is completely reasonable, but I think it should be documented.
- The autocomplete is guided by what is written in `tlm.txt` and will write out breaking code if used.
  - This raises another point that the actual item name is different than what is used in the screens which is confusing.
- If an array item has an index in the name of the item, it is not intuitive as to how to index it.
  - For example if my item name is the following:
  `APPEND_ARRAY_ITEM MYARRAY[12] 8 UINT 80 "Array data"`
  In order to index this I would need to do the following:
  `LABELVALUE MYTARGET STATUS MYARRAY[5][12]`

Some further comments:
- What are the allowed characters in an item name?
  - The reason I ask this is because as it stands it appears that the syntax is changing in the screens thus one needs to be careful as to what they name their items.
- A suggestion I have is to possibly add a new flag to the `VALUE` widget to pass in an index. Maybe something like the following (just spitballing):
  - `LABELVALUE MYTARGET STATUS MYARRAY[12] INDEX:8`
- Another suggestion would be to only index the `ARRAY_ITEM` type. There are still issues though with how the indexing though that I mentioned above.

jmthomas, 2024-08-02:
We made these changes to support Array items as a first class citizen. Using brackets in an item name has always been a bit weird since we already have array items.

1. We should document this better and I'll leave this open until we do
2. Autocomplete should be updated to support this new escape syntax
3. We should enforce not allowing brackets in Array items
4. Parameter naming is enforced in [config_parser](https://github.com/OpenC3/cosmos/blob/632f65cf22ecc6e26ad5eee89bc5b39c8fa47bc0/openc3/lib/openc3/config/config_parser.rb#L254). This is not allowed: `__`, `[[`, `]]`, ` ` (space), starting with `}`, and ending with `_`. Everything else is allowed but in general I would recommend avoiding special characters.

In order to support array items in both screens and TlmGrapher we need to keep the current implementation.

stephen-ritter, 2024-08-05:
Sounds good. I think the long term solution for me would be to start using the `ARRAY_ITEM`, but I will adjust the screens with the `[[n]]` fix for now. Thank you for the tip by the way. I also wanted to add an update as I was cleaning up some screens that this only applies to the screens that use widgets that derive from `VWidget.js`. `LED`, `SETTING`, `LINEGRAPH`, etc appear to want what was defined in the `tlm.txt`.

stephen-ritter, 2024-08-06:
As a further update, it appears that the Limits Monitor breaks as well if you have a telemetry item with indexing in its name. I assume the reason for this is because the Limits Monitor is using widgets derived from `VWidget.js`. I don't know a way of escaping those without editing the source code.

jmthomas, 2025-12-16:
I can confirm the bug with Limits Monitor and created #2644